knarfS/smuview #1: Does not compile

littleyoda, 2017-12-06:
Hi

I tried to compile smuview, but I got the following error:

`[  7%] Building CXX object CMakeFiles/smuview.dir/src/devices/device.cpp.o
/home/sven/devel/sigrok/smuview/src/devices/device.cpp: In member function ‘void sv::devices::Device::feed_in_meta(std::shared_ptr<sigrok::Meta>)’:
/home/sven/devel/sigrok/smuview/src/devices/device.cpp:347:8: error: ‘SR_CONF_UNDER_VOLTAGE_CONDITION_THRESHOLD’ was not declared in this scope
   case SR_CONF_UNDER_VOLTAGE_CONDITION_THRESHOLD:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sven/devel/sigrok/smuview/src/devices/device.cpp:347:8: note: suggested alternative: ‘SR_CONF_OVER_VOLTAGE_PROTECTION_THRESHOLD’
   case SR_CONF_UNDER_VOLTAGE_CONDITION_THRESHOLD:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        SR_CONF_OVER_VOLTAGE_PROTECTION_THRESHOLD
CMakeFiles/smuview.dir/build.make:335: die Regel für Ziel „CMakeFiles/smuview.dir/src/devices/device.cpp.o“ scheiterte
make[2]: *** [CMakeFiles/smuview.dir/src/devices/device.cpp.o] Fehler 1
`

I didn't find SR_CONF_UNDER_VOLTAGE_CONDITION_THRESHOLD in libsigrok.

Are you using a modified sigrok?

knarfS, 2017-12-07:
Hi Yoda,

yes, i'm also improving libsigrok in some points.
You can find my libsigrok git here: https://github.com/knarfS/libsigrok/tree/wip (Branch "wip")
I'm confident, that most of my changes will be merged into sigrok upstream.

Hth,
Frank

littleyoda, 2017-12-07:
Compiling this libsigrok, I run into the next error

`  CXX      bindings/cxx/classes.lo
In file included from bindings/cxx/include/libsigrokcxx/libsigrokcxx.hpp:1077:0,
                 from bindings/cxx/classes.cpp:26:
bindings/cxx/include/libsigrokcxx/enums.hpp:24:32: error: redeclaration of ‘const sigrok::LogLevel* const sigrok::LogLevel::NONE’
  static const LogLevel * const NONE;
                                ^~~~
bindings/cxx/include/libsigrokcxx/enums.hpp:12:32: note: previous declaration ‘const sigrok::LogLevel* const sigrok::LogLevel::NONE’
  static const LogLevel * const NONE;
                                ^~~~
`

Thanks for helping.

knarfS, 2017-12-08:
At the moment I'm preparing my changes in libsigrok, to be mergable. This should be done in the next few days. I suggest you try to compile again, when I'm done. I will inform you here.

Though this error doesn't look like, it has anything to do with my changes, it is probably a good thing to tidy up my changes :)

littleyoda, 2017-12-08:
You are right about the error:
I run into this problem:
https://sigrok.org/bugzilla/show_bug.cgi?id=547

knarfS, 2017-12-13:
I disabled the parts where SR_CONF_UNDER_VOLTAGE_CONDITION_THRESHOLD is used, until it is part of the official libsigrok.

closed 2017-12-13 by knarfS