knarfS/smuview #3: Missing Voltage and Ampere Graph

littleyoda, 2017-12-29:
I tested the master branch against my KD3005P Power Supply.

1. The Voltage & Ampere Graph is missing

2. If I try to change the Voltage or Ampere Limit, smuview crashs with the following error:
```
sr: korad-kaxxxxp: Can't set measurable parameter.
Caught exception: generic/unspecified error
*** Error in ./smuview: double free or corruption (out): 0x000055e86ac47300 ***
```
Using the sig-cli-tool, it works without a problem:
`sigrok-cli -d korad-kaxxxxp:conn=/dev/ttyACM0 --set --config voltage_target=17`

[image]
[image]

knarfS, 2018-01-05:
Hi Sven,

Can you please post the output of "sigrok-cli --driver korad-kaxxxxp:conn=???? --show -l 5"
I had a quick look into the korad-kaxxxxp driver and found several problems.

1. I don't know exactly why "Can't set measurable parameter." occurs, but i think someone (smuview? sigrok?) is trying to set a value to the read only VOLTAGE or CURRENT keys. I have to look into this one.

2. The driver has only one channel "CH1", but should have two: "V" and "I". And in korad_kaxxxxp_receive_data() the corresponding channel should be the only one in analog.meaning->channels list. This is the reason why you see no plots.

3. You may have a locale problem in korad_kaxxxxp_set_value() with snprintf() (I didn't looked into this).

4. The driver is not thread safe. SmuView uses the main thread to send the control commands to the device and a separate thread per device to receive the data from the device. The sending and receiving parts in the driver have to be synchronized.

If you would like to fix this problems, i can talk you through the changes you have to make. Non of the changes should be a big deal, except no. 4 maybe... :)

littleyoda, 2018-01-05:
Hi

attached the logfile ...
[sigrok-cli-show.txt](https://github.com/knarfS/smuview/files/1607451/sigrok-cli-show.txt)

Thanks

knarfS, 2018-01-05:
The output confirms point 2 of my previous post.
Should be no problem to fix.

knarfS, 2018-01-05:
Here is a branch with some fixes:
https://github.com/knarfS/libsigrok/tree/korad-kaxxxxp

littleyoda, 2018-01-05:
Thanks for fixing the channel-Problem with the korad-kaxxxp-Hardware. It works now. Voltage and Ampere is now shown on the UI.

Attached the requested logfile for the smuview.
[smuview-log-for-korad-kaxxxxp.txt](https://github.com/knarfS/smuview/files/1607906/smuview-log-for-korad-kaxxxxp.txt)

Thanks

knarfS, 2018-01-07:
I added some patches to my branch, that should address points 1. and 4.
The code is totally untested, but at least that gives you a starting point.

I'm pretty sure, that there will be a locale problem (point 3 of my list), but we will see :)

littleyoda, 2018-01-07:
Hi

You are right about the locale problem:
```
sr: [00:20.497603] hwdriver: sr_config_set(): key 30030 (voltage_target) sdi 0x5624dd2a2ee0 cg NULL -> 5.2000000000000002
sr: [00:20.575960] korad-kaxxxxp: Sleeping for processing 78253 usec
sr: [00:20.575996] korad-kaxxxxp: korad_kaxxxxp_set_value(): cmd = VSET1:%05.2f
sr: [00:20.576007] korad-kaxxxxp: korad_kaxxxxp_set_value(): msg = VSET1:05,20
sr: [00:20.576015] korad-kaxxxxp: Sending 'VSET1:05,20'.
sr: [00:20.576039] serial: Wrote 11/11 bytes.
```

And the output of sigrok-cli:
```
sr: [00:03.575401] hwdriver: sr_config_set(): key 30030 (voltage_target) sdi 0x55f510a47130 cg NULL -> 17.0
sr: [00:03.628420] korad-kaxxxxp: Sleeping for processing 52900 usec
sr: [00:03.628457] korad-kaxxxxp: korad_kaxxxxp_set_value(): cmd = VSET1:%05.2f
sr: [00:03.628476] korad-kaxxxxp: korad_kaxxxxp_set_value(): msg = VSET1:17.00
sr: [00:03.628483] korad-kaxxxxp: Sending 'VSET1:17.00'.
sr: [00:03.628520] serial: Wrote 11/11 bytes.
```

Full Logfile for smuview attached:
[korad-kaxxxxp-de_de.txt](https://github.com/knarfS/smuview/files/1609661/korad-kaxxxxp-de_de.txt)

littleyoda, 2018-01-08:
With your last commit, setting the voltage and ampere works also. Thanks a lot.

knarfS, 2018-01-08:
New patches added to the korad-kaxxxxp driver.

littleyoda, 2018-01-12:
Attached the requested tests.
One for sigrok-cli and one smuview

[smuview.log](https://github.com/knarfS/smuview/files/1626733/smuview.log)
[test-kd3005p.log](https://github.com/knarfS/smuview/files/1626810/test-kd3005p.log)

closed 2018-02-25 by knarfS