knarfS/smuview #6: Can't add new signal

Wurstnase, 2019-03-21:
Just tried the latest github version, because I need to change my Korad driver slightly.
(Normally answering with KORADKA3005V2.0<-68>)

Finally smuview found the device but stops while trying to connect.

```
BaseDevice::close(): Trying to close device  "Korad KA3005P (/dev/ttyACM0)"
sr: korad-kaxxxxp: Want max 22 bytes.
Configurable::init(): Init  "Korad KA3005P"  - key  "Regulation"
Configurable::init(): Init  "Korad KA3005P"  - key  "Over Current Protection Enabled"
Configurable::init(): Init  "Korad KA3005P"  - key  "Over Voltage Protection Enabled"
Configurable::init(): Init  "Korad KA3005P"  - key  "Enabled"
Configurable::init(): Init  "Korad KA3005P"  - key  "Current Limit"
Configurable::init(): Init  "Korad KA3005P"  - key  "Current"
Configurable::init(): Init  "Korad KA3005P"  - key  "Voltage Target"
Configurable::init(): Init  "Korad KA3005P"  - key  "Voltage"
Init channel  "V" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init channel  "I" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init analog signal  "V [V DC]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init analog signal  "I [A DC]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init channel  "" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init analog signal  "P [W]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init channel  "" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init analog signal  "R [Ω]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init channel  "" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init analog signal  "Wh [Wh]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init channel  "" , channel_start_timestamp =  "2019.03.21 16:39:10.688"
Init analog signal  "Ah [Ah]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Start aquisition for  "Korad KA3005P" ,  aquisition_start_timestamp_ =  "2019.03.21 16:39:10.688"
Init analog signal  "I [A]" , signal_start_timestamp_ =  "2019.03.21 16:39:10.688"
Can't add new signal  "I [A]" to fixed channel  "I"
BaseSignal::~BaseSignal():  "I [A]"
HardwareChannel::push_sample_sr_analog():  "I"  - No signal found:  "I [A DC]"
[1]    20700 segmentation fault  sudo ./smuview
```

knarfS, 2019-03-21:
Hey Wurstnase,

thanks for submitting this bug.
I've added a "feature" that prevents adding new signals to fixed channels like the voltage and current channels of power supplies. Maybe not the best idea :)

I've removed this feature in commit 52bef770357d9bfc3a1128380c322fb1c9932e58 and I've added a patch to my pending branch of patches for libsigrok (see https://github.com/knarfS/libsigrok/commit/763c652b60e1239852059f93f47e26916a83eed4).

--Frank

Wurstnase, 2019-03-21:
Perfect. Just test it and it's working. Only my monitor is too small for initial settings and I need to move any subwindow first.

A small patch probably for your collection. This will compare only for the length of the current model-id instead of the complete reply. So in case of KORADRA3005PV2.0\x01 is not necessary anymore. My korad answer with 0xBC instead of 0x01. Maybe other Korad will answer differently in the future.
```
diff --git a/src/hardware/korad-kaxxxxp/api.c b/src/hardware/korad-kaxxxxp/api.c
index eac10b80..11680b65 100644
--- a/src/hardware/korad-kaxxxxp/api.c
+++ b/src/hardware/korad-kaxxxxp/api.c
@@ -52,9 +52,6 @@ static const struct korad_kaxxxxp_model models[] = {
 		"VELLEMANLABPS3005DV2.0", 1, {0, 31, 0.01}, {0, 5, 0.001}},
 	{KORAD_KA3005P, "Korad", "KA3005P",
 		"KORADKA3005PV2.0", 1, {0, 31, 0.01}, {0, 5, 0.001}},
-	/* Sometimes the KA3005P has an extra 0x01 after the ID. */
-	{KORAD_KA3005P_0X01, "Korad", "KA3005P",
-		"KORADKA3005PV2.0\x01", 1, {0, 31, 0.01}, {0, 5, 0.001}},
 	{KORAD_KD3005P, "Korad", "KD3005P",
 		"KORAD KD3005P V2.0", 1, {0, 31, 0.01}, {0, 5, 0.001}},
 	{KORAD_KD3005P_V20_NOSP, "Korad", "KD3005P",
@@ -126,7 +123,7 @@ static GSList *scan(struct sr_dev_driver *di, GSList *options)
 	sr_dbg("Received: %d, %s", i, reply);
 	model_id = -1;
 	for (i = 0; models[i].id; i++) {
-		if (!strcmp(models[i].id, reply))
+		if (!strncmp(models[i].id, reply, strlen(models[i].id)))
 			model_id = i;
 	}
 	if (model_id < 0) {
diff --git a/src/hardware/korad-kaxxxxp/protocol.h b/src/hardware/korad-kaxxxxp/protocol.h
index 34c6465a..2178a5a0 100644
--- a/src/hardware/korad-kaxxxxp/protocol.h
+++ b/src/hardware/korad-kaxxxxp/protocol.h
@@ -35,7 +35,6 @@ enum {
 	VELLEMAN_PS3005D,
 	VELLEMAN_LABPS3005D,
 	KORAD_KA3005P,
-	KORAD_KA3005P_0X01,
 	KORAD_KD3005P,
 	KORAD_KD3005P_V20_NOSP,
 	RND_320K30PV,
```

knarfS, 2019-03-29:
Your patch was rejected because of potential side effects with other model ids, but your model id with the extra 0xBC byte is now in upstream: https://github.com/sigrokproject/libsigrok/commit/1c0e1baaaaf0314f49470fb49a4f5e6ddc0ba1a1

Maybe you could open another bug here with screenshots, your monitor settings and so on. This way I can improve the use of the available display space.

closed 2019-03-29 by knarfS