knarfS/smuview #9: CRC error RDTech DPS

robotfreak, 2019-04-05:
I got  a CRC Error when trying to connect a RDTech DPS Power supply:

```
./SmuView-0.0.3-x86_64.AppImage 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Configurable::init(): Init  "A0"  - key  "Offset"
Configurable::init(): Init  "A0"  - key  "Measured Quantity"
Configurable::init(): Init  "A0"  - key  "Amplitude"
Configurable::init(): Init  "A0"  - key  "Pattern Mode"
Configurable::init(): Init  "A1"  - key  "Offset"
Configurable::init(): Init  "A1"  - key  "Measured Quantity"
Configurable::init(): Init  "A1"  - key  "Amplitude"
Configurable::init(): Init  "A1"  - key  "Pattern Mode"
Configurable::init(): Init  "A2"  - key  "Offset"
Configurable::init(): Init  "A2"  - key  "Measured Quantity"
Configurable::init(): Init  "A2"  - key  "Amplitude"
Configurable::init(): Init  "A2"  - key  "Pattern Mode"
Configurable::init(): Init  "A3"  - key  "Offset"
Configurable::init(): Init  "A3"  - key  "Measured Quantity"
Configurable::init(): Init  "A3"  - key  "Amplitude"
Configurable::init(): Init  "A3"  - key  "Pattern Mode"
Configurable::init(): Init  "Analog"  - key  "Offset"
Configurable::init(): Init  "Analog"  - key  "Amplitude"
Configurable::init(): Init  "Logic"  - key  "Pattern Mode"
Configurable::init(): Init  "Demo device"  - key  "Trigger Match"
Configurable::init(): Init  "Demo device"  - key  "Averaging Samples"
Configurable::init(): Init  "Demo device"  - key  "Averaging"
Configurable::init(): Init  "Demo device"  - key  "Capture Ratio"
Configurable::init(): Init  "Demo device"  - key  "Samplerate"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Regulation"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Current Protection Threshold"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Current Protection Active"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Voltage Protection Threshold"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Voltage Protection Active"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Enabled"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Current Limit"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Current"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Voltage Target"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Voltage"
Init channel  "V" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init channel  "I" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init channel  "P" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init analog signal  "V [V DC]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Init analog signal  "I [A DC]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Init analog signal  "P [W]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Init channel  "" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init analog signal  "R [Ω]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Init channel  "" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init analog signal  "Wh [Wh]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Init channel  "" , channel_start_timestamp =  "2019.04.05 21:51:39.753"
Init analog signal  "Ah [Ah]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Start aquisition for  "RDTech DPS5005" ,  aquisition_start_timestamp_ =  "2019.04.05 21:51:39.753"
Caught exception: generic/unspecified error
BaseDevice::close(): Trying to close device  "RDTech DPS5005 v14"
Init analog signal  "P [W DC]" , signal_start_timestamp_ =  "2019.04.05 21:51:39.753"
Warning: Adding new signal  "P [W DC]" to fixed channel  "P"
HardwareChannel::push_sample_sr_analog():  "P"  - No signal found:  "P [W DC]"
sr: modbus_serial: CRC error (0x0000 vs 0xFDC3).
BaseDevice::close(): Device closed  "RDTech DPS5005 v14"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"
```

knarfS, 2019-04-05:
Hi robofreak,

looks like a bug in the sigrok rdtech-dps driver, or more specific in the modbus CRC check (modbus_serial_rtu.c -> modbus_serial_rtu_read_end()).
Could you start SmuView with the "-l 5" argument, connect the device and attach the log. Maybe the log contains more info about the CRC check.
I don't own one of the RDTech power supplies and therefore I can't debug the problem myself.

Furthermore the driver is missing a mutex. That's a generic fix I can add, when the CRC check is working.

littleyoda, 2019-04-06:
had the same error with sigrok-cli. So it is a bug in the driver. 
I send one or two hours investigation this bug, but without any real success

robotfreak, 2019-04-06:
Thank you, for your fast response.
This was my first guess too, that this is a sigrok specific error. But with the sigrok-cli (v0.7.1) the power supply is detected and works.

```
./sigrok-cli-0.7.1-x86_64.AppImage -d rdtech-dps:conn=/dev/ttyUSB0 --scan
The following devices were found:
rdtech-dps - RDTech DPS5005 v14 with 3 channels: V I P
peter@peter-labor:~/Downloads$ sudo ./sigrok-cli-0.7.1-x86_64.AppImage -d rdtech-dps:conn=/dev/ttyUSB0 --show
Driver functions:
    Power supply
Scan options:
    conn
    serialcomm
    modbusaddr
rdtech-dps - RDTech DPS5005 v14 with 3 channels: V I P
Supported configuration options:
    continuous: on, off
    limit_samples: 0 (current)
    limit_time: 0 (current)
    voltage: 5.000000
    voltage_target: 5.000000
    current: 0.000000
    current_limit: 5.100000
    enabled: on (current), off
    regulation: CV (current)
    ovp_active: on, off (current)
    ovp_threshold: 51.000000
    ocp_active: on, off (current)
    ocp_threshold: 5.200000
peter@peter-labor:~/Downloads$ sudo ./sigrok-cli-0.7.1-x86_64.AppImage -d rdtech-dps:conn=/dev/ttyUSB0 --continous
Unknown option --continous
peter@peter-labor:~/Downloads$ sudo ./sigrok-cli-0.7.1-x86_64.AppImage -d rdtech-dps:conn=/dev/ttyUSB0 --continuos
Unknown option --continuos
peter@peter-labor:~/Downloads$ sudo ./sigrok-cli-0.7.1-x86_64.AppImage -d rdtech-dps:conn=/dev/ttyUSB0 --continuous
FRAME-BEGIN
V: 4.990 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END
FRAME-BEGIN
V: 5.000 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END
FRAME-BEGIN
V: 5.000 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END
FRAME-BEGIN
V: 5.000 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END
FRAME-BEGIN
V: 5.000 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END
FRAME-BEGIN
V: 5.000 V DC
I: 0.0 mA DC
P: 0 mW DC
FRAME-END

`````
Here is the log from smuview. I see it is linked again sigrok v0.6. So maybe this error is already fixed in newer sigrok versions:

```
./SmuView-0.0.3-x86_64.AppImage -l 5
[... 323 lines cut ...]
sr: [01:11.979403] session: bus: Received SR_DF_FRAME_END packet.
sr: [01:11.979477] serial: Wrote 1/1 bytes.
sr: [01:11.979538] serial: Wrote 5/5 bytes.
sr: [01:11.979603] serial: Wrote 2/2 bytes.
sr: [01:11.979648] session: Stopping.
sr: [01:11.979687] hwdriver: rdtech-dps: Stopping acquisition.
sr: [01:11.979727] session: bus: Received SR_DF_END packet.
sr: [01:11.979793] session: fd_source_finalize: key 0x7f18f400d7d0
sr: [01:11.979870] session: Stopped.
sr: [01:11.980146] device: rdtech-dps: Closing device instance.
sr: [01:12.436264] serial: Read 1/1 bytes.
sr: [01:12.436349] serial: Read 1/1 bytes.
sr: [01:12.436406] serial: Read 5/5 bytes.
sr: [01:12.436475] serial: Read 2/2 bytes.
sr: [01:12.436513] modbus_serial: CRC error (0x0000 vs 0xFDC3).
sr: [01:12.436582] serial: Wrote 1/1 bytes.
sr: [01:12.436647] serial: Wrote 8/8 bytes.
sr: [01:12.436734] serial: Wrote 2/2 bytes.
sr: [01:12.436794] serial: Read 1/1 bytes.
sr: [01:12.436828] serial: Closing serial port /dev/ttyUSB0.
BaseDevice::close(): Device closed  "RDTech DPS5005 v14"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"

```

littleyoda, 2019-04-06:
interesting! 
I got the error normally after 10 to 20 min runing sigrok

knarfS, 2019-04-07:
I've looked into the driver code, but without the device there is not much i can do. Unfortunately, the logs you attached doesn't help either.
I'm going to add a bunch of log outputs to the relevant parts and post an AppImage here asap. Maybe I get a clue somehow :)

Perhaps the error occurs so fast with SmuView because of the missing mutex? I'll add the mutex in the test AppImage too.
But that sounds like a sporadic error. Maybe the device sends a wrong checksum sometimes? Hopefully more log output will tell...

@littleyoda : You have a Korad psu, don't you? It would be great, if you could confirm the thing with the unreliable buttons and check if my fix is working (issue #7 ).

littleyoda, 2019-04-07:
My Modell of the Korad psu do not have these buttons. Sorry, can not test it :-(

robotfreak, 2019-04-09:
I have done some further investigation. When thesigrok-cli is running in log level 5 I can see the crc error occuring, when the USB connection has been closed. 
So this explains the different behaviours  between the sigrok-cli and the smuview app. The smuview app closes the connection after init. The sigrok-cli keeps the connection open until you close the program itself. 

```
sr: [01:11.521536] serial: Read 1/1 bytes.
sr: [01:11.521612] serial: Read 1/1 bytes.
sr: [01:11.521664] serial: Read 7/7 bytes.
sr: [01:11.521717] serial: Read 2/2 bytes.
sr: [01:11.521764] session: bus: Received SR_DF_FRAME_BEGIN packet.
sr: [01:11.521817] session: bus: Received SR_DF_ANALOG packet (1 samples).
sr: [01:11.521973] session: bus: Received SR_DF_ANALOG packet (1 samples).
sr: [01:11.522077] session: bus: Received SR_DF_ANALOG packet (1 samples).
Init analog signal  "P [W DC]" , signal_start_timestamp_ =  "2019.04.06 11:37:25.049"
Warning: Adding new signal  "P [W DC]" to fixed channel  "P"
HardwareChannel::push_sample_sr_analog():  "P"  - No signal found:  "P [W DC]"
sr: [01:11.522426] session: bus: Received SR_DF_FRAME_END packet.
sr: [01:11.522497] serial: Wrote 1/1 bytes.
sr: [01:11.522559] serial: Wrote 5/5 bytes.
sr: [01:11.522621] serial: Wrote 2/2 bytes.
sr: [01:11.978881] serial: Read 1/1 bytes.
sr: [01:11.978951] serial: Read 1/1 bytes.
sr: [01:11.979005] serial: Read 7/7 bytes.
sr: [01:11.979065] serial: Read 2/2 bytes.
sr: [01:11.979107] session: bus: Received SR_DF_FRAME_BEGIN packet.
sr: [01:11.979159] session: bus: Received SR_DF_ANALOG packet (1 samples).
sr: [01:11.979258] session: bus: Received SR_DF_ANALOG packet (1 samples).
sr: [01:11.979341] session: bus: Received SR_DF_ANALOG packet (1 samples).
sr: [01:11.979403] session: bus: Received SR_DF_FRAME_END packet.
sr: [01:11.979477] serial: Wrote 1/1 bytes.
sr: [01:11.979538] serial: Wrote 5/5 bytes.
sr: [01:11.979603] serial: Wrote 2/2 bytes.
sr: [01:11.979648] session: Stopping.
sr: [01:11.979687] hwdriver: rdtech-dps: Stopping acquisition.
sr: [01:11.979727] session: bus: Received SR_DF_END packet.
sr: [01:11.979793] session: fd_source_finalize: key 0x7f18f400d7d0
sr: [01:11.979870] session: Stopped.
sr: [01:11.980146] device: rdtech-dps: Closing device instance.
sr: [01:12.436264] serial: Read 1/1 bytes.
sr: [01:12.436349] serial: Read 1/1 bytes.
sr: [01:12.436406] serial: Read 5/5 bytes.
sr: [01:12.436475] serial: Read 2/2 bytes.
sr: [01:12.436513] modbus_serial: CRC error (0x0000 vs 0xFDC3).
sr: [01:12.436582] serial: Wrote 1/1 bytes.
sr: [01:12.436647] serial: Wrote 8/8 bytes.
sr: [01:12.436734] serial: Wrote 2/2 bytes.
sr: [01:12.436794] serial: Read 1/1 bytes.
sr: [01:12.436828] serial: Closing serial port /dev/ttyUSB0.
BaseDevice::close(): Device closed  "RDTech DPS5005 v14"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"

```

knarfS, 2019-04-09:
Hi robotfreak,
are you able to compile libsigrok/sigrok-cli/smuview by yourself? I've added the missing mutex and a whole bunch of debug output for the RDTech psu to my branch "patches": https://github.com/knarfS/libsigrok/commits/patches
Otherwise I can create an AppImage at the end of this week.

You don't need to add "-l 5" for the log output this time.
I don't know if we will see some useful info in the outputs, but i hope so.

robotfreak, 2019-04-10:
Hi knarfS,
ok, I build the libsigrok (branch patches) and smuview (master) by myself.
Here is the log output:

```
./smuview
Configurable::init(): Init  "A0"  - key  "Offset"
Configurable::init(): Init  "A0"  - key  "Measured Quantity"
Configurable::init(): Init  "A0"  - key  "Amplitude"
Configurable::init(): Init  "A0"  - key  "Pattern Mode"
Configurable::init(): Init  "A1"  - key  "Offset"
Configurable::init(): Init  "A1"  - key  "Measured Quantity"
Configurable::init(): Init  "A1"  - key  "Amplitude"
Configurable::init(): Init  "A1"  - key  "Pattern Mode"
Configurable::init(): Init  "A2"  - key  "Offset"
Configurable::init(): Init  "A2"  - key  "Measured Quantity"
Configurable::init(): Init  "A2"  - key  "Amplitude"
Configurable::init(): Init  "A2"  - key  "Pattern Mode"
Configurable::init(): Init  "A3"  - key  "Offset"
Configurable::init(): Init  "A3"  - key  "Measured Quantity"
Configurable::init(): Init  "A3"  - key  "Amplitude"
Configurable::init(): Init  "A3"  - key  "Pattern Mode"
Configurable::init(): Init  "Analog"  - key  "Offset"
Configurable::init(): Init  "Analog"  - key  "Amplitude"
Configurable::init(): Init  "Logic"  - key  "Pattern Mode"
Configurable::init(): Init  "Demo device"  - key  "Trigger Match"
Configurable::init(): Init  "Demo device"  - key  "Averaging Samples"
Configurable::init(): Init  "Demo device"  - key  "Averaging"
Configurable::init(): Init  "Demo device"  - key  "Capture Ratio"
Configurable::init(): Init  "Demo device"  - key  "Samplerate"
sr: rdtech-dps: probe_device(): -1-
sr: rdtech-dps: rdtech_dps_get_model_version(): -1-
sr: modbus_serial: modbus_serial_rtu_read_data(): function_code = 0x3
sr: modbus_serial: modbus_serial_rtu_read_begin(): slave_addr = 0x1 (1)
sr: modbus_serial: modbus_serial_rtu_read_begin(): -1- modbus->crc = 0x807E
sr: modbus_serial: modbus_serial_rtu_read_begin(): -2- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): len = 5, maxlen = 5
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x4
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x5
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x6
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x7
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x8
sr: modbus_serial: modbus_serial_rtu_read_data(): -1- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): -2- modbus->crc = 0x58EF
sr: modbus_serial: modbus_serial_rtu_read_end(): crc = 0x58EF (2)
sr: rdtech-dps: rdtech_dps_get_model_version(): -2- ret = 0
sr: rdtech-dps: probe_device(): -2- ret = 0
Configurable::init(): Init  "RDTech DPS5005"  - key  "Regulation"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Current Protection Threshold"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Current Protection Active"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Voltage Protection Threshold"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Voltage Protection Active"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Enabled"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Current Limit"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Current"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Voltage Target"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Voltage"
sr: rdtech-dps: dev_open(): -1-
sr: rdtech-dps: rdtech_dps_set_reg(): -1-
sr: modbus_serial: modbus_serial_rtu_read_data(): function_code = 0x10
sr: modbus_serial: modbus_serial_rtu_read_begin(): slave_addr = 0x1 (1)
sr: modbus_serial: modbus_serial_rtu_read_begin(): -1- modbus->crc = 0x807E
sr: modbus_serial: modbus_serial_rtu_read_begin(): -2- modbus->crc = 0xEC01
sr: modbus_serial: modbus_serial_rtu_read_data(): len = 4, maxlen = 4
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x0
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x1
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x2
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x3
sr: modbus_serial: modbus_serial_rtu_read_data(): -1- modbus->crc = 0xEC01
sr: modbus_serial: modbus_serial_rtu_read_data(): -2- modbus->crc = 0xC8E1
[... 224 lines cut ...]
sr: rdtech-dps: dev_close(): -1-
sr: modbus_serial: modbus_serial_rtu_read_data(): function_code = 0x3
sr: modbus_serial: modbus_serial_rtu_read_begin(): slave_addr = 0x1 (1)
sr: modbus_serial: modbus_serial_rtu_read_begin(): -1- modbus->crc = 0x807E
sr: modbus_serial: modbus_serial_rtu_read_begin(): -2- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): len = 5, maxlen = 5
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x6
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x7
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x8
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x9
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0xA
sr: modbus_serial: modbus_serial_rtu_read_data(): -1- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): -2- modbus->crc = 0x3C72
sr: modbus_serial: modbus_serial_rtu_read_end(): crc = 0x0 (2)
sr: modbus_serial: CRC error (0x0000 vs 0x3C72).
sr: rdtech-dps: dev_close(): -2- ret = -10
sr: rdtech-dps: dev_close(): -3-
sr: rdtech-dps: rdtech_dps_set_reg(): -1-
sr: rdtech-dps: rdtech_dps_set_reg(): -2- ret = -1
sr: rdtech-dps: dev_close(): -4- ret = -1
BaseDevice::close(): Device closed  "RDTech DPS5005 v14"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"

```

robotfreak, 2019-04-10:
Hi knarfS,

success. I got it working. The only change I made, is in the rdtech_dps_capture_start function. Setting devc->expecting_registers to 0, instead of 2. Not sure if this is the right way to fix it, but for me it works. Now the device_close function will never read any pending data.

```
SR_PRIV int rdtech_dps_capture_start(const struct sr_dev_inst *sdi)
{
	struct dev_context *devc;
	struct sr_modbus_dev_inst *modbus;
	int ret;

	devc = sdi->priv;
	modbus = sdi->conn;

	g_mutex_lock(&devc->rw_mutex);
	sr_err("rdtech_dps_capture_start(): -1-");
	if ((ret = sr_modbus_read_holding_registers(modbus, REG_UOUT, 3, NULL)) == SR_OK)
		devc->expecting_registers = 0; 
	sr_err("rdtech_dps_capture_start(): -2- ret = %i", ret);
	g_mutex_unlock(&devc->rw_mutex);
	return ret;
}
```

knarfS, 2019-04-10:
The CRC error was a red herring and only a subsequent error... :(
The actual error hasn't any interesting log output, but at least I can narrow it down.

I don't think your fix solved the actual error, though the expecting_registers thing doesn't make much sense the way it is implemented...

I updated my branch with more log output in the interesting parts. Would be great if you could attach a log without your patch and one with.
Sorry, but I rebased my patch branch, you have to "git fetch --all && git reset --hard origin/patches".

robotfreak, 2019-04-10:
Now I get a memory access violation error with or without my patch the log output looks the same:

.```
/smuview
Configurable::init(): Init  "A0"  - key  "Offset"
Configurable::init(): Init  "A0"  - key  "Measured Quantity"
Configurable::init(): Init  "A0"  - key  "Amplitude"
Configurable::init(): Init  "A0"  - key  "Pattern Mode"
Configurable::init(): Init  "A1"  - key  "Offset"
Configurable::init(): Init  "A1"  - key  "Measured Quantity"
Configurable::init(): Init  "A1"  - key  "Amplitude"
Configurable::init(): Init  "A1"  - key  "Pattern Mode"
Configurable::init(): Init  "A2"  - key  "Offset"
Configurable::init(): Init  "A2"  - key  "Measured Quantity"
Configurable::init(): Init  "A2"  - key  "Amplitude"
Configurable::init(): Init  "A2"  - key  "Pattern Mode"
Configurable::init(): Init  "A3"  - key  "Offset"
Configurable::init(): Init  "A3"  - key  "Measured Quantity"
Configurable::init(): Init  "A3"  - key  "Amplitude"
Configurable::init(): Init  "A3"  - key  "Pattern Mode"
Configurable::init(): Init  "Analog"  - key  "Offset"
Configurable::init(): Init  "Analog"  - key  "Amplitude"
Configurable::init(): Init  "Logic"  - key  "Pattern Mode"
Configurable::init(): Init  "Demo device"  - key  "Trigger Match"
Configurable::init(): Init  "Demo device"  - key  "Averaging Samples"
Configurable::init(): Init  "Demo device"  - key  "Averaging"
Configurable::init(): Init  "Demo device"  - key  "Capture Ratio"
Configurable::init(): Init  "Demo device"  - key  "Samplerate"
sr: rdtech-dps: probe_device(): -1-
sr: rdtech-dps: rdtech_dps_get_model_version(): -1-
sr: modbus: sr_modbus_read_holding_registers(): address = 0xB, nb_registers = 2
sr: modbus: sr_modbus_read_holding_registers(): request = 0x0 0x0 0x0 0x0 0x0
sr: modbus: sr_modbus_read_holding_registers(): request = 0x3 0x0 0xB 0x0 0x2
sr: modbus_serial: modbus_serial_rtu_send(): slave_addr = 0x1 (1)
sr: modbus_serial: modbus_serial_rtu_send(): ret = 1
sr: modbus_serial: modbus_serial_rtu_send(): buffer_size = 5
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x3
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x4
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x5
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x6
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x7
sr: modbus_serial: modbus_serial_rtu_send(): ret = 5
sr: modbus_serial: modbus_serial_rtu_send(): -1- crc = 0x807E (2)
sr: modbus_serial: modbus_serial_rtu_send(): -1- crc = 0xC9B5 (2)
sr: modbus_serial: modbus_serial_rtu_send(): ret = 2
sr: modbus: sr_modbus_read_holding_registers(): ret = 0
sr: modbus: sr_modbus_read_holding_registers(): registers = 0x7FB1 (32689)
sr: modbus_serial: modbus_serial_rtu_read_begin(): slave_addr = 0x1 (1), ret = 1
sr: modbus_serial: modbus_serial_rtu_read_data(): function_code = 0x3, ret = 1
sr: modbus_serial: modbus_serial_rtu_read_begin(): -1- modbus->crc = 0x807E
sr: modbus_serial: modbus_serial_rtu_read_begin(): -2- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): len = 5, maxlen = 5
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x4
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x5
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x6
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x7
sr: modbus_serial: modbus_serial_rtu_read_data(): buf = 0x8
sr: modbus_serial: modbus_serial_rtu_read_data(): -1- modbus->crc = 0x2140
sr: modbus_serial: modbus_serial_rtu_read_data(): -2- modbus->crc = 0x58EF
sr: modbus_serial: modbus_serial_rtu_read_end(): crc = 0x58EF (2)
sr: modbus: sr_modbus_read_holding_registers(): ret = 0
sr: modbus: sr_modbus_read_holding_registers(): -1-
sr: modbus: sr_modbus_read_holding_registers(): -2-
sr: modbus: sr_modbus_read_holding_registers(): -3-
sr: modbus: sr_modbus_read_holding_registers(): registers = 0x8D13 (36115)
sr: modbus: sr_modbus_read_holding_registers(): -4-
sr: rdtech-dps: rdtech_dps_get_model_version(): -2- ret = 0
sr: rdtech-dps: probe_device(): -2- ret = 0
Configurable::init(): Init  "RDTech DPS5005"  - key  "Regulation"
Configurable::init(): Init  "RDTech DPS5005"  - key  "Over Current Protection Threshold"
[... 53 lines cut ...]
Init analog time signal  "Wh [Wh]" , signal_start_timestamp_ =  "2019.04.10 22:24:20.203"
Init channel  "" , channel_start_timestamp =  "2019.04.10 22:24:20.203"
Init analog base signal  "Ah [Ah]"
Init analog time signal  "Ah [Ah]" , signal_start_timestamp_ =  "2019.04.10 22:24:20.203"
sr: rdtech-dps: dev_acquisition_start(): -1-
sr: rdtech-dps: rdtech_dps_capture_start(): -1-
sr: modbus: sr_modbus_read_holding_registers(): address = 0x2, nb_registers = 3
sr: modbus: sr_modbus_read_holding_registers(): request = 0x0 0x0 0x0 0x0 0x0
sr: modbus: sr_modbus_read_holding_registers(): request = 0x3 0x0 0x2 0x0 0x3
sr: modbus_serial: modbus_serial_rtu_send(): slave_addr = 0x1 (1)
sr: modbus_serial: modbus_serial_rtu_send(): ret = 1
sr: modbus_serial: modbus_serial_rtu_send(): buffer_size = 5
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x3
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x4
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x5
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x6
sr: modbus_serial: modbus_serial_rtu_send(): buf = 0x7
sr: modbus_serial: modbus_serial_rtu_send(): ret = 5
sr: modbus_serial: modbus_serial_rtu_send(): -1- crc = 0x807E (2)
sr: modbus_serial: modbus_serial_rtu_send(): -1- crc = 0xBA4 (2)
sr: modbus_serial: modbus_serial_rtu_send(): ret = 2
sr: modbus: sr_modbus_read_holding_registers(): ret = 0
Speicherzugriffsfehler (Speicherabzug geschrieben)

```

knarfS, 2019-04-10:
Ups, my bad.
I updated my branch.

robotfreak, 2019-04-11:
ok, here are the logfiles for the updates branch with and without my patch. 
[smuview-patches3-log.txt](https://github.com/knarfS/smuview/files/3070548/smuview-patches3-log.txt)
[smuview-mypatches3-log.txt](https://github.com/knarfS/smuview/files/3070549/smuview-mypatches3-log.txt)

knarfS, 2019-04-14:
Development moved to https://github.com/knarfS/libsigrok/tree/patches2

Both tests failed, when trying to get the ovp_active config key. The code for getting the config key looks OK, as far as I can tell.

Could you try to get the key with sigrok-cli (replace conn string with your settings):
`sigrok-cli -d rdtech-dps:conn=/dev/ttyUSB0 --get ovp_active -l 5`
This will probably fail too:
`sigrok-cli -d rdtech-dps:conn=/dev/ttyUSB0 --get ocp_active -l 5`
 
Next thing I can do is deactivate this two config key and then testing if there are other config keys that are failing.
But without the device it is very hard to tell what's wrong with those config keys...

robotfreak, 2019-04-14:
Here are the log files for ovp and ocp active config key:
[smuview-patches2-ovp.txt](https://github.com/knarfS/smuview/files/3077817/smuview-patches2-ovp.txt)
[smuview-patches2-ocp.txt](https://github.com/knarfS/smuview/files/3077818/smuview-patches2-ocp.txt)

knarfS, 2019-04-15:
I think i got it! It was hopefully only a timeout that was set to low.
Please attach another log output, so i can properly adjust the timeout.
I've moved development to https://github.com/knarfS/libsigrok/tree/patches2

robotfreak, 2019-04-15:
Here are the log files. The  serial_read_blocking time varies from 33...124 for ovp, 19...91 for ocp.
[smuview-patches2-ovp2.txt](https://github.com/knarfS/smuview/files/3082093/smuview-patches2-ovp2.txt)
[smuview-patches2-ocp2.txt](https://github.com/knarfS/smuview/files/3082095/smuview-patches2-ocp2.txt)

knarfS, 2019-04-15:
The important duration is 1st serial_read_blocking() in modbus_serial_rtu_read_begin() which is around 450ms. Pretty close to the former timeout of 500ms.

But the most important question: Is SmuView working now?

robotfreak, 2019-04-16:
No, smuview don't work. Here is the log:
[smuview-patches2-log.txt](https://github.com/knarfS/smuview/files/3086509/smuview-patches2-log.txt)

knarfS, 2019-04-16:
Yeah, but my assumption was right ;) However the timeout of 1000ms was still to low...
I increased the timeout to 5s, that should hopefully be enough.

robotfreak, 2019-04-16:
Cool, now smuview is working. Thank you so much knarfS

knarfS, 2019-04-16:
Could you attach the log output, so I can adjust the timeouts and make a pull request to libsigrok.

robotfreak, 2019-04-17:
Damn, this is too bad. Yesterday smuview is working, today not. Here is the logfile from today. I have not got one from yesterday.
[smuview-notok.txt](https://github.com/knarfS/smuview/files/3091403/smuview-notok.txt)

knarfS, 2019-04-17:
It's again the timeout. Seems like sometimes 5s isn't enough. Puhhh....

Could you play around with the timeout by yourself? You'll find it in src/modbus/modbus_serial_rtu.c, line 148. I'd start with 20000ms (THAT should be enough) and then look at the time in the log ("modbus_serial_rtu_read_begin(): 1st serial_read_blocking() time =") and adjust it down. When SmuView is running stable, start to change the output voltage, current, toggle the output, etc. to stress the power supplies lil' CPU and check if the duration of the serial_read_blocking() is rising...

robotfreak, 2019-04-22:
Sorry, but I don't get the timeout working. Even with 40s isn't enough. At the moment the chances are 50:50 to start smuview succesfully. If smuview starts successfully, the blocking time is always less than 5s.
[smuview-log-not-ok.txt](https://github.com/knarfS/smuview/files/3104579/smuview-log-not-ok.txt)
[smuview-log2.txt](https://github.com/knarfS/smuview/files/3104583/smuview-log2.txt)

knarfS, 2019-05-01:
Sorry for the late reply.
I pushed a possible workaround to my patches2 branch (rebased!): The sr_modbus_read_holding_registers() function is now called up to 3 times if it fails. i also decreased the timeout back to reasonable 600ms.
Let's see if it's working now...

robotfreak, 2019-05-02:
No worry. Now it is working like a charm. Phantastic job!
[smuview-log2.txt](https://github.com/knarfS/smuview/files/3139897/smuview-log2.txt)

knarfS, 2019-05-22:
I moved the patches for the RDTech power supplies back to my branch "patches" (https://github.com/knarfS/libsigrok/commits/patches).
I refined the commits (a few debug outputs are left, but will be removed before I make a pull request).

I also added meta packages: When the regulation (CV/CC) or the output state changes or one of the protections (OVP/OCP) is triggered, that will be displayed in SmuView.

I would be happy, if you could test my changes again and append a log file with log level 5 (./smuview -l 5)

robotfreak, 2019-05-29:
Here are the log files (first failed connection, second successful connection):
[smuview-patches-nok.txt](https://github.com/knarfS/smuview/files/3234489/smuview-patches-nok.txt)
[smuview-patches-ok.txt](https://github.com/knarfS/smuview/files/3234490/smuview-patches-ok.txt)

knarfS, 2020-02-18:
Most of the issues should be fixed in libsigrok since 16.12.2019 and are now available in the SmuView 0.0.4 binaries.

closed 2020-02-18 by knarfS

17acres, 2020-03-20:
> Most of the issues should be fixed in libsigrok since 16.12.2019 and are now available in the SmuView 0.0.4 binaries.

Can anyone confirm that the x86_64  appimage from 0.0.4 is working for the DPS? I tried on both Windows and Ubuntu and experienced the same CRC error with the files from the latest release. I've confirmed that the DPS works properly using RD Tech's software on Windows, so I don't think it's a problem on my hardware. I tried building from source on Ubuntu and ran into python issues using either the method from the SmuView manual or the Sigrok Building from Source page.

knarfS, 2020-03-21:
> Can anyone confirm that the x86_64 appimage from 0.0.4 is working for the DPS? I tried on both Windows and Ubuntu and experienced the same CRC error with the files from the latest release. I've [...]

I've implemented a workaround in the sigrok driver for the DPS devices some time ago (see sigrokproject/libsigrok@aff2094). Because I don't own one of those PSUs, I can't do a proper investigation of that problem.
What I can tell so far: Sometimes the device doesn't respond to the "read holding registers" modbus command. The workaround tries to read the registers up to 3 times.
Somebody with that devices has to do some deeper investigation and maybe debug the protocol the vendor is using in its software and tweak the sigrok driver.   

> I tried building from source on Ubuntu and ran into python issues using either the method from the SmuView manual or the Sigrok Building from Source page.

Can you give some details about that (CMake and/or make output)? Probably a missing dependency or something similar. Maybe open a new issue.