knarfS/smuview #13: build error Ubuntu 18.04

jwd996, 2020-02-18:
building trunk source via
cmake -DCMAKE_CXX_STANDARD=14
(I had to get cmake to remove -std=c++11 flag)

does not succed:
/usr/local/qwt-6.1.3/lib/libqwt.so: undefined reference to `qt_version_tag@Qt_5.10'
//home/jw/local/Qt/5.10.1/gcc_64/lib/libQt5OpenGL.so.5: undefined reference to `QRandomGenerator::_fillRange(void*, void*)@Qt_5'
//home/jw/local/Qt/5.10.1/gcc_64/lib/libQt5PrintSupport.so.5: undefined reference to `QPdfEngine::setPdfVersion(QPdfEngine::PdfVersion)@Qt_5_PRIVATE_API'
/usr/local/qwt-6.1.3/lib/libqwt.so: undefined reference to `operator delete(void*, unsigned long)@Qt_5'
//home/jw/local/Qt/5.10.1/gcc_64/lib/libQt5PrintSupport.so.5: undefined reference to `QTime::toString(QStringView) const@Qt_5'
//home/jw/local/Qt/5.10.1/gcc_64/lib/libQt5OpenGL.so.5: undefined reference to `QRandomGenerator64::global()@Qt_5'

jwd996, 2020-02-18:
Found a way to use QtCreator to build. QtCreator comes with Qt5.14.
But to get it working, I had to add the line
**set( CMAKE_CXX_STANDARD  14)**

between 
**include(GNUInstallDirs)
project(smuview)**
in CMakeFiles.txt

otherwise cmake would have added "-std=gnu++11" to compiler flags, which might come from some configuration of my system. To make the build process more robust on various build environments, I would recommend this change.

closed 2020-02-18 by jwd996

knarfS, 2020-02-19:
SmuView and the libs in `/external` only require C++11. But there was a glitch in the matrix which was fixed in commit 60b8d20.
I'll add a `set(CMAKE_CXX_STANDARD 11)` to the `CMakeLists.txt` to be more explicit.

The other problem is, that you are using a custom Qt 5.10.1 for compiling and linking, but when executing, the binary is trying to use your systems Qt 5.9.5. That's based on the info here and from the mikrocontroller forum (https://www.mikrocontroller.net/topic/490066#6149038):
```
Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50a01)
``` 
That does not work, because for Qt libraries, minor releases are only backwards compatible (binary and source).

But you could try to add your custom Qt libraries path to the `LD_LIBRARY_PATH` environment variable (not tested!):
```
$ env LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/qt-5.10.1 ./smuview
```