knarfS/smuview #14: Build Issues on Ubuntu 18.04

17acres, 2020-03-21:
**Describe the bug**
When building either manually as described at https://sigrok.org/wiki/Linux#Building_.28manually.29 or using the script at https://knarfs.github.io/doc/smuview/0.0.4/manual.html#_linux , the build fails with Python errors. PulseView and the rest of the Sigrok suite install fine.
**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps for manual installation of all Sigrok tools at https://sigrok.org/wiki/Linux#Building_.28manually.29

or 

2. Follow the building from source directions here https://knarfs.github.io/doc/smuview/0.0.4/manual.html#_linux 

**Expected behavior**
The application should install without error.

**Screenshots or log**
Using the script, I first get this error:
```
  GEN      bindings/python/timestamp
destdir=''; /usr/bin/python ../bindings/python/setup.py VERSION='0.6.0' CC='g++' CXX='g++' CFLAGS='-g -O2 -Wall -Wextra -I/usr/include/python2.7 -I/usr/include/x86_64-linux-gnu/python2.7 -I/usr/include/pygobject-3.0 -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include' CXXFLAGS='-g -O2 -Wall -Wextra -I/usr/include/python2.7 -I/usr/include/x86_64-linux-gnu/python2.7 -I/usr/include/pygobject-3.0 -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include' LDADD='-lpython2.7 -lglibmm-2.4 -lgobject-2.0 -lglib-2.0 -lsigc-2.0' --quiet install ${destdir:+"--root=$destdir"} \
	--prefix "/home/dannooo/sr" --exec-prefix "/home/dannooo/sr"
TEST FAILED: /home/dannooo/sr/lib/python2.7/site-packages/ does NOT support .pth files
error: bad install directory or PYTHONPATH

You are attempting to install a package to a directory that is not
on PYTHONPATH and which Python does not read ".pth" files from.  The
installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /home/dannooo/sr/lib/python2.7/site-packages/

and your PYTHONPATH environment variable currently contains:

    '/home/dannooo/sr/lib/python3.5/site-packages'

Here are some of your options for correcting the problem:

* You can choose a different installation directory, i.e., one that is
  on PYTHONPATH or supports .pth files

* You can add the installation directory to the PYTHONPATH environment
  variable.  (It must then also be on PYTHONPATH whenever you run
  Python and want to use the package(s) you are installing.)

* You can set up the installation directory to support ".pth" files by
  using one of the approaches described here:

  https://setuptools.readthedocs.io/en/latest/easy_install.html#custom-installation-locations

Please make the appropriate changes for your system and try again.

```

I tried editing the script to set PYTHONPATH to point at Python 2.7 (I know this isn't a reasonable thing to do but it is worth a shot). This got me past this stage and into the "make" phase of the build process.  Then, I reach the following error later in compilation. This is the same error as what happens following the manual installation steps:

```
[100%] Linking CXX executable smuview
CMakeFiles/smuview.dir/src/python/bindings.cpp.o: In function `pybind11::detail::string_caster<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, false>::cast(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pybind11::return_value_policy, pybind11::handle) [clone .isra.840]':
/home/dannooo/sr/sigrok-util/cross-compile/linux/build/smuview/external/pybind11/include/pybind11/cast.h:1268: undefined reference to `PyUnicode_DecodeUTF8'
CMakeFiles/smuview.dir/src/python/bindings.cpp.o: In function `pybind11::detail::type_caster<char, void>::cast(char const*, pybind11::return_value_policy, pybind11::handle) [clone .constprop.1894]':
/home/dannooo/sr/sigrok-util/cross-compile/linux/build/smuview/external/pybind11/include/pybind11/pytypes.h:190: undefined reference to `_Py_NoneStruct'
CMakeFiles/smuview.dir/src/python/bindings.cpp.o: In function `init_Session(pybind11::module&)':
/home/dannooo/sr/sigrok-util/cross-compile/linux/build/smuview/external/pybind11/include/pybind11/pytypes.h:190: undefined reference to `_Py_NoneStruct'
...
...

```
There's maybe thousands of lines of the same "undefined reference" errors.

**Enviroment :**
 - SmuView commit 13db44a
 - OS: Ububtu 18.04.4LTS

knarfS, 2020-04-10:
The first part of the bug (wrong PYTHONPATH when building libsigrok) can be solved (as you did) by setting the correct PYTHONPATH (documented [here](https://sigrok.org/wiki/Building#TEST_FAILED:_....2Flib.2Fpython2.7.2Fsite-packages.2F_does_NOT_support_.pth_files)).
An other possibility is to set the default python interpreter to python3, something like:
```
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.4 1
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.6 2
sudo update-alternatives --set python /usr/bin/python3.6
```

The second part (error when linking SmuView) should be fixed with commit a9a51b6.

Please let me know, if this is fixed for you.

17acres, 2020-04-10:
With the python fix, the build works (I still can't get the RDTech DPS to communicate but I can look into that more myself).

Is there any reason PYTHONPATH should be set to Python 3.5 for most installs? [sigrok-cross-linux-smuview](https://github.com/knarfS/sigrok-util/blob/smuview/cross-compile/linux/sigrok-cross-linux-smuview) sets PYTHONPATH to Python 3.5, while sigrok-cross-linux sets it to Python 2.7. So you can't just manually set PYTHONPATH as explained on [the help page you linked](https://sigrok.org/wiki/Building#TEST_FAILED:_....2Flib.2Fpython2.7.2Fsite-packages.2F_does_NOT_support_.pth_files) then run the compile/install script, since it overwrites PYTHONPATH. I just changed the line in sigrok-cross-linux-smuview to the below and it works:
`# The path where the libsigrok Python bindings will be installed.
PYPATH=$PREFIX/lib/python2.7/site-packages`

Thanks for the fix

knarfS, 2020-04-10:
> Is there any reason PYTHONPATH should be set to Python 3.5 for most installs?

You must set `PYPATH` (and this way indirectly `PYTHONPATH`) in `sigrok-cross-linux-smuview` to match your default python version. `PYPATH` is only used for building the libsigrok python bindings (which aren't used for SmuView anyways).
I prefer to use python 3.x, because python 2.7 is finally dead :) Therefore `PYPATH` is set to 3.5 in `sigrok-cross-linux-smuview`.
I probably have to add a sentence about this variable to the manual. 

SmuView will automatically use python 3.x, no matter what `PYPATH` is set to.

Hope this helps.

closed 2020-04-10 by knarfS

knarfS, 2020-04-10:
> With the python fix, the build works (I still can't get the RDTech DPS to communicate but I can look into that more myself).

There is a recent commit in libsigrok (https://github.com/sigrokproject/libsigrok/commit/69c5d959e75b9bf9689be39da0dd8abd6078d89a) that may fix this issue.

17acres, 2021-01-15:
Can now confirm that both the new Windows release (0.0.5) and the Linux version (at least when built from source) work fine with the RDTech DPS series power supplies.

knarfS, 2021-01-16:
Thanks for letting me know!