knarfS/smuview #17: AppImage: Bluetooth LE devices, like the 121gw

fasmide, 2020-05-24:
I'm so delighted by the availability of an AppImage build of SmuView as there are just too many ways to get a current build of sigrok and smuview on ubuntu :)

**Describe the bug**
When running the app image and selecting, in my case, the EEVBlog 121gw multimeter, smuview begins to scan for bluetooth devices but it seems to fail only for LE devices. It does find my Nvidia Shield (a tv box) and even my neighbor's Sony TV 

**To Reproduce**
Steps to reproduce the behavior:
1. Run SmuView.AppImage
2. Click Add Device
3. Select "EEVBLOG 121gw"
4. Choose Serial Port

.... No LE devices found, only regular Bluetooth devices ... 

**Expected behavior**
I expected to select my multimeter right next to the pc :)

**Screenshots or log**
I won't bother you with screenshots, but the terminal does make a few interesting statements:

When running as my regular user:
```
$ ./SmuView-0.0.4-x86_64.AppImage 
set LE scan params: Operation not permitted
set LE scan params: Operation not permitted
```

When running as root (though not recommended)
```
21:14 $ sudo ./SmuView-0.0.4-x86_64.AppImage 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
set LE scan params: Input/output error
```

**Enviroment (please complete the following information):**
SmuView-0.0.4-x86_64.AppImage 
Ubuntu 19.10
Linux 5.3.0-45-generic

**Additional context**
The manual states some udev rules which need to be applied in order to make the app image function as advertised - I had a look at them and I ... well I don't know but at least my gut feeling tells me they have nothing to do with this - and I haven't applied them

knarfS, 2020-05-25:
I'll do some tests next weekend, when I have my 121GW back.
But AFAICR you had to pair the 121GW first (hcitool? gatttool? KDE Bluetooth GUI thingy?) and after that it was visible in the connection dialog. But that was to long ago to remember and I could be wrong.

If that works, I'm interested how the 121GW works for you. Hint: For me, the 121 only connects successfully every 1 out of 10 tries. :)

fasmide, 2020-05-26:
Well, it's not the most stable thing in the world - even the official android app seems a bit dodgy as dave himself would put it :)

Anyways I did some more digging and I found that even using `hcitool` directly I would get similar errors:
```
21:46 $ hcitool lescan
Set scan parameters failed: Operation not permitted
21:46 $ sudo hcitool lescan
Set scan parameters failed: Input/output error
```

So I naturally updated my ubuntu install to a more recent 20.04 LTS only to find the exact same behavior. 
However tools like the `blueman-applet` are able to detect (and even "pair" with) the meter - which by the way makes no difference for smuview or sigrok-cli as far as I can tell - but at least something is talking

... I wouldn't worry too much about the scanning errors - though annoying - I don't think they are smuview's fault. It must be my specific setup or maybe faulty ubuntu default settings/permissions

But, now that I have a more recent `sigrok-cli` i tried to use the cli directly, passing the mac address of the meter to the driver, and voila! - its works ... I would say, 1 out of 3 times at least :)

```
22:00 $ sigrok-cli -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 --continuous
main: 40.3 °C
sub: 22.4 °C
 ..... snip .... 
22:01 $ sigrok-cli -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 --continuous
sr: serial: Didn't find a valid packet (read 0 bytes).
No devices found.
22:01 $ sigrok-cli -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 --continuous
sr: serial: Didn't find a valid packet (read 0 bytes).
No devices found.
22:01 $ sigrok-cli -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 --continuous
sr: serial: Didn't find a valid packet (read 0 bytes).
No devices found.
22:02 $ sigrok-cli -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 --continuous
main: 32.8 °C
sub: 23.0 °C
.... strip ...
```

Maybe the meter needs to timeout the Bluetooth "connection" in order to accept a new?
Anyways back to smuview - now it should only be a matter of passing the same command line parameters, but sadly no:

```
22:06 $ ./SmuView-0.0.4-x86_64.AppImage --loglevel 5 -D -d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-B0-76 
sr: [00:00.000000] log: libsigrok loglevel set to 5.
Settings: "/home/fas/.config/sigrok/SmuView.conf" format QSettings::NativeFormat
sr: [00:00.001798] backend: libsigrok 0.6.0-git-8491cf7/4:0:0.
sr: [00:00.003096] backend: Libs: glib 2.48.2 (rt: 2.64.2/6402:2), libzip 1.0.1, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.23.11397 API 0x01000104, hidapi 0.8.0-rc1, bluez 5.37, libftdi 1.2.
sr: [00:00.003107] backend: Host: x86_64-pc-linux-gnu, little-endian.
sr: [00:00.003119] backend: SCPI backends: TCP, RPC, serial, USBTMC.
sr: [00:00.003126] backend: Firmware search paths:
sr: [00:00.003159] backend:  - /home/fas/.local/share/sigrok-firmware
sr: [00:00.003167] backend:  - /home/frank/smuview/build/deps/share/sigrok-firmware
sr: [00:00.003174] backend:  - /usr/share/i3/sigrok-firmware
sr: [00:00.003181] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.003188] backend:  - /usr/share/sigrok-firmware
sr: [00:00.003195] backend:  - /var/lib/snapd/desktop/sigrok-firmware
sr: [00:00.003202] backend:  - /var/lib/snapd/desktop/sigrok-firmware
sr: [00:00.003814] backend: Sanity-checking all drivers.
sr: [00:00.003830] backend: Sanity-checking all input modules.
sr: [00:00.003838] backend: Sanity-checking all output modules.
sr: [00:00.003850] backend: Sanity-checking all transform modules.
sr: [00:00.064233] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi (nil) cg NULL -> [uint32 20000, 20001]
sr: [00:00.064328] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi (nil) cg NULL -> [uint32 20000, 20001]
sr: [00:00.064350] serial: Opening serial port 'bt/ble122/88-6B-0F-81-B0-76' (flags 1).
sr: [00:00.064385] bt-bluez: BLE connect, remote addr 88:6B:0F:81:B0:76
sr: [00:01.344719] bt-bluez: BLE start notify
sr: [00:01.344752] serial: Parsing parameters from "115200/8n1".
sr: [00:01.344814] serial: Setting serial parameters on port bt/ble122/88-6B-0F-81-B0-76.
sr: [00:01.344823] serial: DBG: serial_set_params() rate 115200, 8n1
sr: [00:01.344871] serial-dmm: Probing serial port bt/ble122/88-6B-0F-81-B0-76.
sr: [00:01.346277] serial: Flushing serial port bt/ble122/88-6B-0F-81-B0-76.
[... 40 lines cut ...]
Init channel  "bar" , channel_start_timestamp =  "2020.05.26 22:06:50.823"
sr: [00:01.652564] session: Creating our own main context.
sr: [00:01.652605] session: Starting.
sr: [00:01.652610] hwdriver: eevblog-121gw: Starting acquisition.
sr: [00:01.652622] session: bus: Received SR_DF_HEADER packet.
Start aquisition for  "EEVblog 121GW" ,  aquisition_start_timestamp_ =  "2020.05.26 22:06:50.823"
sr: [00:01.752858] bt-bluez: read() len 1, type 0x13 (write response)
```
Nothing more shows up in the logging after this - smuview does show up and presents a nice interface for the meter but never updates:

[image]

My best guess would be these lines from smuview's log output

```
sr: [00:01.507514] bt-bluez: BLE disconnect
sr: [00:01.507521] bt-bluez: BLE close
sr: [00:01.507553] bt-bluez: BLE close
```

If smuview closes the BLE connection - it's just gonna have to wait what seems like ... maybe a minute? before trying to make a new connection (for the actual data acquisition)

I did try the Stop button and wait to see if starting it again after some amount of time would do the trick - no luck

Let me know if your testing goes different :)

knarfS, 2020-06-02:
Good news: I was able to test my 121GW over the weekend and with the latest libsigrok git master everything looks good!

The problem of not updating the GUI (I had this bug as well) is now gone! :partying_face:
Though, when closing SmuView and restart it immediately (with `-d eevblog-121gw:conn=bt/ble122/88-6B-0F-81-AA-C3`) the device fails to connect. I had to wait 1-2 seconds to reconnect it successfully. No big deal, I guess.

Also it isn't necessary to pair the 121GW with a 3rd party tool beforehand, I just started SmuView with the appropriate `-d` option and the dmm connected flawlessly.
But the device isn't listed in the Serial Port dropdown list of the connection dialog and I had to type in the connection string (`bt/ble122/88-6B-0F-81-AA-C3`) by hand.
When starting SmuView as root, the device is listed in the connection dialog.

I'm afraid you have to compile libsigrok and smuview by hand for now :)
A new release (and AppImage) is in the making, but that will take a few more weeks.

fasmide, 2020-08-03:
That's great news! - Sorry I haven't reached back in a while - I thought to my self: no worries I can wait on the next build - but when is it coming? ;)

I did try building sigrok and ... well ... I don't know - its just kind of tedious, requiring all kinds of weird and sometimes deprecated versions of packages... 

I also tried building the smuview appimage inside a container to ensure my workstation wasn't left with all these requirements, however, it seems the appimage-thing needs special filesystem permissions to properly pack up the final image :/

anyways - I'm looking forward to the next build :)

knarfS, 2020-08-09:
Building libsigrok and SmuView should be pretty straight forward when you follow the wiki instructions or alternatively use the script described in the SmuView manual ([Chapter 2.1](https://knarfs.github.io/doc/smuview/0.0.4/manual.html#_linux), Section 2). But yes, there are quite some dependencies (see [wiki](https://sigrok.org/wiki/Linux#Building)) you have to install before building :)
Building the AppImage by yourself though is not trivial (This could be used as a starting point: [release.txt](https://github.com/knarfS/smuview/blob/master/stuff/release.txt))!

But there is hope: I'm almost done with the features I want to implement for the upcoming version 0.0.5. Main feature is that the GUI will remember your settings and layout.
My goal is to merge my changes to the main branch in the next 1-2 weeks and to release a new version in ~2 months.

knarfS, 2021-01-11:
0.0.5 released with an actual libsigrok and working 121gw driver.

closed 2021-01-11 by knarfS

ludrol, 2026-09-10:
I have come across the same problem with mooshimeter (bluetooth multimeter). I got the continuous github appimage. 

```
sudo  ./SmuView-0.0.6-git-0927de3-x86_64.appimage  --version 
SmuView 0.0.6-git-0927de3

sudo ./SmuView-0.0.6-git-0927de3-x86_64.appimage 
set LE scan enable: Input/output error
```

sigrok-cli works in container based on archlinux. appimage fails due to insufficient permissions in container.