knarfS/smuview #18: cannot/no longer compile due to deprecated declarations

lintweaker, 2020-05-30:
**Describe the bug**
Fresh compile with the latest git version fails due to deprecated declarations

**To Reproduce**
Steps to reproduce the behavior:
Compile fresh from git source (on Fedora 30 at least)

**Expected behavior**
Nice compile :-)

**Screenshots or log**
`[ 90%] Building CXX object CMakeFiles/smuview.dir/src/ui/widgets/lcddisplay.cpp.o
[ 91%] Building CXX object CMakeFiles/smuview.dir/src/ui/widgets/monofontdisplay.cpp.o
/home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp: In member function ‘virtual void sv::ui::widgets::MonoFontDisplay::update_value_widget_dimensions()’:
/home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp:137:47: error: ‘int QFontMetrics::width(const QString&, int) const’ is deprecated: Use QFontMetrics::horizontalAdvance [-Werror=deprecated-declarations]
  137 |  value_label_->setFixedWidth(metrics.width(str));
      |                                               ^
In file included from /usr/include/qt5/QtGui/QFontMetrics:1,
                 from /home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp:22:
/usr/include/qt5/QtGui/qfontmetrics.h:108:9: note: declared here
  108 |     int width(const QString &, int len = -1) const;
      |         ^~~~~
/home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp: In member function ‘virtual void sv::ui::widgets::MonoFontDisplay::update_unit_widget_dimensions()’:
/home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp:159:46: error: ‘int QFontMetrics::width(const QString&, int) const’ is deprecated: Use QFontMetrics::horizontalAdvance [-Werror=deprecated-declarations]
  159 |  unit_label_->setFixedWidth(metrics.width(str));
      |                                              ^
In file included from /usr/include/qt5/QtGui/QFontMetrics:1,
                 from /home/kramer/Documents/keysight-u1242b/sigrok/smuview/smuview/src/ui/widgets/monofontdisplay.cpp:22:
/usr/include/qt5/QtGui/qfontmetrics.h:108:9: note: declared here
  108 |     int width(const QString &, int len = -1) const;
      |         ^~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/smuview.dir/build.make:1680: CMakeFiles/smuview.dir/src/ui/widgets/monofontdisplay.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:163: CMakeFiles/smuview.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
`

**Enviroment (please complete the following information):**
 - SmuView version / git commit : 4b5f36779fd201ee19908ea56fe7d07f08c1fd05
 - self compiled: gcc (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2)
 - OS: Fedora 30 x86_64

**Additional context**

knarfS, 2020-06-02:
There is already a PR over at Gitlab that should fix this issue: https://gitlab.com/MrRabbit/smuview-qt5-5.14.2/-/commits/master/
If you are keen, give these patches a try and let me know if this works for you.

lintweaker, 2020-06-02:
Was able to successfully compile the version from gitlab!
https://gitlab.com/MrRabbit/smuview-qt5-5.14.2

closed 2020-06-02 by lintweaker

knarfS, 2020-06-02:
Great! I'm planing to merge the patch in the near future.