knarfS/smuview #20: Add a new power supply

fgervais, 2020-07-02:
I just received a new IT6922A power supply and I'm looking at adding support for it in smuview.

My current state is that I just crash coursed myself on SCPI, I gathered the SCPI command manual of my unit and I poked my unit for the first time:

```
fgervais@fgervais-System-Product-Name:~$ echo "*IDN?" > /dev/usbtmc2
fgervais@fgervais-System-Product-Name:~$ cat /dev/usbtmc2
ITECH Ltd., IT6922A, ******************,  1.10-1.06
```

I'd like to get a better understanding of the work required to add support for it in smuview.

fgervais, 2020-07-06:
I see now. I think I need to add the support to libsigrok and smuview should be able to use it.

I'll give this a try.

closed 2020-07-06 by fgervais

knarfS, 2020-07-07:
Yes, that is right. SmuView is based on libsigrok, where all the device drivers are implemented.
If the IT6922A has a SCPI base protocol (looks like it does), please have a look at the scpi-pps driver. It should be pretty straight forward to add basic functionality for your device.
Also a good source for help is the #sigrok Frenode IRC channel. Maybe you have to wait some time to get an answer, but most of the developers hang around there.