knarfS/smuview #22: Unknown model ID 'KORAD KA3005P V5.5' detected, aborting while trying to connect to KA3005P.

renearts, 2020-10-02:
**Describe the bug**
I have a brand new KA3005P which I am trying to connect to my Ubuntu 20.04 PC via USB. The device connects to /dev/ttyACM0. When running SmuView (AppImage, 0.0.4-x86_64) I get the following error:
```
/SmuView-0.0.4-x86_64.AppImage 
set LE scan params: Operation not permitted
set LE scan params: Operation not permitted
sr: korad-kaxxxxp: Unknown model ID 'KORAD KA3005P V5.5' detected, aborting.
```

I also tried:
```
/SmuView-0.0.4-x86_64.AppImage -d korad-kaxxxxp:conn=/dev/ttyACM0 
sr: korad-kaxxxxp: Unknown model ID 'KORAD KA3005P V5.5' detected, aborting.
set LE scan params: Operation not permitted
set LE scan params: Operation not permitted
sr: korad-kaxxxxp: Unknown model ID 'KORAD KA3005P V5.5' detected, aborting.
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"
```
I can press scan for devices in the GUI, but there is no device showing up. I have selected the correct Korad device in the dropdown.

**Enviroment (please complete the following information):**
 - SmuView version 0.0.4 x86_64 AppImage
 - OS: Ubuntu 20.04

BayBuilder, 2020-10-11:
I have the exact same problem. Interesting that there is a several second pause between reading the serial response and the rest of the output.

Linux Mint 19.3 Cinnamon (based on Ubuntu Bionic 18.04.3 LTS)
SmuView-0.0.4-x86_64.AppImage
Korad KA3005P Programmable DC Power Supply connected via USB

Press "Add Device" button
Select "Korad KAxxxxP (korad-kaxxxxp)" driver
Select "Serial Port" [ /dev/ttyACM0 (USB Virtual COM - 002914100246) is filled in automatically]
Press "Scan for devices using driver above"
No devices appear under "Step 4: Select the device"

Terminal comand:
./SmuView-0.0.4-x86_64.AppImage --loglevel 5

The following appears in the terminal after pressing "Scan for devices using driver above":

>sr: [00:16.571349] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi (nil) cg NULL -> [uint32 20000, 20001]
sr: [00:16.571420] serial: Opening serial port '/dev/ttyACM0' (flags 1).
sr: [00:16.572211] serial: Parsing parameters from "9600/8n1".
sr: [00:16.572255] serial: Setting serial parameters on port /dev/ttyACM0.
sr: [00:16.572289] serial: DBG: serial_set_params() rate 9600, 8n1
sr: [00:16.572305] serial: Flushing serial port /dev/ttyACM0.
sr: [00:16.572314] korad-kaxxxxp: Want max 34 bytes.
sr: [00:16.572318] korad-kaxxxxp: Sending '*IDN?'.
sr: [00:16.572337] serial: Wrote 5/5 bytes.
sr: [00:16.617443] serial: Read 27/34 bytes.
sr: [00:16.662651] serial: Read 3/7 bytes.
sr: [00:21.106259] korad-kaxxxxp: Received: 'KORAD KA3005P V5.5 SN:276*****'.
sr: [00:21.106329] korad-kaxxxxp: Received: 0, KORAD KA3005P V5.5 SN:276*****
sr: [00:21.106363] korad-kaxxxxp: Unknown model ID 'KORAD KA3005P V5.5' detected, aborting.
sr: [00:21.106390] hwdriver: Scan found 0 devices (korad-kaxxxxp).

30 bytes are always returned but they are sometimes split differently:

>sr: [00:25.785698] serial: Wrote 5/5 bytes.
sr: [00:25.830822] serial: Read 28/34 bytes.
sr: [00:25.876005] serial: Read 2/6 bytes.

The rest of the terminal output is always identical.

BayBuilder, 2020-10-11:
Somehow managed to compile from source and everything is working! Must be something in the .AppImage file that isn't right.

SmuView 0.0.4-git-6e934f0
Linux Mint 19.3 Cinnamon
Korad KA3005P Programmable DC Power Supply connected via USB

knarfS, 2020-10-11:
The Korad KA3005P with the firmware version 5.5 was added in libsigrok with commit https://github.com/sigrokproject/libsigrok/commit/8b58a519e4d9e972f2cfc2f05ab2fc2c9f3fd591 some time ago. The current AppImage does contain an older version of libsigrok.
At the moment, you have to build libsigrok and SmuView from source if you want support for that power supply.

I'm working on a new SmuView release (updated AppImage), hopefully this year.

renearts, 2021-01-11:
Thanks so much for the update; I just downloaded and tested it and it works brilliantly! 
If there is any way I can support you or the project please let me know, e.g. buy you a cup of coffee or a pint of beer :)

knarfS, 2021-01-11:
In addition (for new unsupported Korad PSUs), there is now the forced detection feature of libsigrok: https://sigrok.org/wiki/Connection_parameters#Forced_detection

closed 2021-01-11 by knarfS