knarfS/smuview #24: Deprecated QT functions used -> compiling fails

maximevince, 2020-11-16:
**Describe the bug**
Unable to compile the latest master (`f5c450e (HEAD -> master, origin/master, origin/HEAD) QCodeEditor: Update to actual knarfS/QCodeEditor fork, commit b03448e`)

**To Reproduce**
Steps to reproduce the behavior:
1. Build according to INSTALL using cmake on ArchLinux

**Expected behavior**
Software to compile successfully.

**Screenshots or log**

```
[ 21%] Building CXX object CMakeFiles/smuview.dir/src/settingsmanager.cpp.o
[ 22%] Building CXX object CMakeFiles/smuview.dir/src/util.cpp.o
/home/vinz/Projects/OpenOtii/smuview/src/util.cpp: In function ‘QString sv::util::format_time_si(const Timestamp&, sv::util::SIPrefix, unsigned int, const QString&, bool)’:
/home/vinz/Projects/OpenOtii/smuview/src/util.cpp:179:9: error: ‘QTextStream& QTextStreamFunctions::forcesign(QTextStream&)’ is deprecated: Use Qt::forcesign [-Werror=deprecated-declarations]
  179 |   ts << forcesign;
      |         ^~~~~~~~~
In file included from /usr/include/qt/QtCore/qdebug.h:49,
                 from /usr/include/qt/QtCore/QDebug:1,
                 from /home/vinz/Projects/OpenOtii/smuview/src/util.cpp:31:
/usr/include/qt/QtCore/qtextstream.h:279:80: note: declared here
  279 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::forcesign") QTextStream &forcesign(QTextStream &s);
      |                                                                                ^~~~~~~~~
/home/vinz/Projects/OpenOtii/smuview/src/util.cpp:179:9: error: ‘QTextStream& QTextStreamFunctions::forcesign(QTextStream&)’ is deprecated: Use Qt::forcesign [-Werror=deprecated-declarations]
  179 |   ts << forcesign;
      |         ^~~~~~~~~
In file included from /usr/include/qt/QtCore/qdebug.h:49,
                 from /usr/include/qt/QtCore/QDebug:1,
                 from /home/vinz/Projects/OpenOtii/smuview/src/util.cpp:31:
/usr/include/qt/QtCore/qtextstream.h:279:80: note: declared here
  279 | Q_CORE_EXPORT QT_DEPRECATED_VERSION_X(5, 15, "Use Qt::forcesign") QTextStream &forcesign(QTextStream &s);
      |                                                                                ^~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/smuview.dir/build.make:224: CMakeFiles/smuview.dir/src/util.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:185: CMakeFiles/smuview.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
make  120,38s user 12,88s system 114% cpu 1:56,66 total

```

**Enviroment (please complete the following information):**
 - SmuView version / git commit: f5c450e
 - Self-compiled version, gcc version 10.2.0 (GCC), QT 5.15.1
 - OS: ArchLinux rolling release

knarfS, 2020-11-19:
Thanks for letting me know. I also updated my local Qt to 5.15.1 :)

closed 2020-11-19 by knarfS

maximevince, 2020-11-21:
Thanks, that works!

commit e4399655 (2020-11-19): Replace obsolete/deprecated Qt methods and use QT_VERSION_CHECK macro everywhere.

Closes #24
Based partly on pulseview commit ae726b70a7ada9a4be5808e00f0c951318479684