knarfS/smuview #25: Error using DMM Agilent U1252A

ekute-dev, 2020-12-01:
## Bug description
DMM Agilent 1252A connected with USB
driver agilent-dmm selected, correct serial post selected

After initialization, smuview seems to try to get data from a second channel that doesn't exist on this meter causing the meter to bip every second or more.

Smuview is sending the serial command 'CONF? @2' which report '*E' that means serial command not understood
same thing with 'FETC? @2'

## Environment 

SmuView-0.0.4-x86_64.AppImage
Debian 10

## log level 5

sr: [00:00.000000] log: libsigrok loglevel set to 5.
Settings: "/home/clement/.config/sigrok/SmuView.conf" format QSettings::NativeFormat
sr: [00:00.003807] backend: libsigrok 0.6.0-git-8491cf7/4:0:0.
sr: [00:00.006556] backend: Libs: glib 2.48.2 (rt: 2.58.3/5803:3), libzip 1.0.1, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.22.11312 API 0x01000104, hidapi 0.8.0-rc1, bluez 5.37, libftdi 1.2.
sr: [00:00.006568] backend: Host: x86_64-pc-linux-gnu, little-endian.
sr: [00:00.006573] backend: SCPI backends: TCP, RPC, serial, USBTMC.
sr: [00:00.006595] backend: Firmware search paths:
sr: [00:00.006608] backend:  - /home/clement/.local/share/sigrok-firmware
sr: [00:00.006612] backend:  - /home/frank/smuview/build/deps/share/sigrok-firmware
sr: [00:00.006618] backend:  - /usr/share/gnome/sigrok-firmware
sr: [00:00.006622] backend:  - /home/clement/.local/share/flatpak/exports/share/sigrok-firmware
sr: [00:00.006627] backend:  - /var/lib/flatpak/exports/share/sigrok-firmware
sr: [00:00.006651] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.006656] backend:  - /usr/share/sigrok-firmware
sr: [00:00.008143] backend: Sanity-checking all drivers.
sr: [00:00.008161] backend: Sanity-checking all input modules.
sr: [00:00.008166] backend: Sanity-checking all output modules.
sr: [00:00.008174] backend: Sanity-checking all transform modules.
sr: [00:00.154657] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.155606] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.174944] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.175702] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.176158] hwdriver: Scan found 0 devices (baylibre-acme).
sr: [00:00.184691] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.185288] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.185809] hwdriver: Scan found 0 devices (brymen-bm857).
sr: [00:00.186113] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.186986] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:00.187278] hwdriver: Scan found 0 devices (cem-dt-885x).
sr: [00:00.187551] hwdriver: Scan found 0 devices (center-309).
sr: [00:00.187895] hwdriver: Scan found 0 devices (colead-slm).
sr: [00:00.188404] hwdriver: Scan found 0 devices (conrad-digi-35-cpu).
sr: [00:00.196394] demo: Generating square pattern.
sr: [00:00.196453] demo: Generating sine pattern.
sr: [00:00.196628] demo: Generating triangle pattern.
sr: [00:00.196774] demo: Generating sawtooth pattern.
sr: [00:00.196847] hwdriver: Scan found 1 devices (demo).
sr: [00:00.198914] hwdriver: Scan found 0 devices (deree-de5000).
sr: [00:00.199286] hwdriver: Scan found 0 devices (digitek-dt4000zc).
sr: [00:00.199598] hwdriver: Scan found 0 devices (eevblog-121gw).
sr: [00:00.206168] hwdriver: Scan found 0 devices (fluke-45).
sr: [00:00.206546] hwdriver: Scan found 0 devices (fluke-dmm).
sr: [00:00.206888] hwdriver: Scan found 0 devices (gmc-mh-1x-2x-rs232).
sr: [00:00.207053] hwdriver: Scan found 0 devices (gmc-mh-2x-bd232).
sr: [00:00.207199] hwdriver: Scan found 0 devices (gwinstek-gpd).
sr: [00:00.212101] hwdriver: Scan found 0 devices (hp-3457a).
sr: [00:00.235398] hwdriver: Scan found 0 devices (hpib-pps).
sr: [00:00.237173] hwdriver: Scan found 0 devices (iso-tech-idm103n).
sr: [00:00.239921] hwdriver: Scan found 0 devices (kecheng-kc-330b).
sr: [00:00.240685] hwdriver: Scan found 0 devices (kern-ew-6200-2nm).
sr: [00:00.241691] hwdriver: Scan found 0 devices (korad-kaxxxxp).
sr: [00:00.246475] hwdriver: Scan found 0 devices (lascar-el-usb).
sr: [00:00.247319] hwdriver: Scan found 0 devices (manson-hcs-3xxx).
sr: [00:00.248193] hwdriver: Scan found 0 devices (mastech-mas345).
[... 332 lines cut ...]
sr: [01:17.668767] serial: Read 1/1 bytes.
sr: [01:17.668822] serial: Read 1/1 bytes.
sr: [01:17.668841] serial: Read 1/1 bytes.
sr: [01:17.668856] serial: Read 1/1 bytes.
sr: [01:17.668864] agilent-dmm: Received '*E'.
sr: [01:17.669058] agilent-dmm: Unknown line '*E'.
sr: [01:17.923037] agilent-dmm: Sending 'FETC? @2'.
sr: [01:17.923106] serial: Wrote 11/11 bytes.
sr: [01:17.956574] serial: Read 1/1 bytes.
sr: [01:17.956632] serial: Read 1/1 bytes.
sr: [01:17.956659] serial: Read 1/1 bytes.
sr: [01:17.956675] serial: Read 1/1 bytes.
sr: [01:17.956689] agilent-dmm: Received '*E'.
sr: [01:17.956893] agilent-dmm: Unknown line '*E'.
sr: [01:18.224008] agilent-dmm: Sending 'FETC? @2'.
sr: [01:18.224104] serial: Wrote 11/11 bytes.
sr: [01:18.244292] serial: Read 1/1 bytes.
sr: [01:18.244345] serial: Read 1/1 bytes.
sr: [01:18.260354] serial: Read 1/1 bytes.
sr: [01:18.260400] serial: Read 1/1 bytes.
sr: [01:18.260409] agilent-dmm: Received '*E'.
sr: [01:18.260574] agilent-dmm: Unknown line '*E'.
sr: [01:18.526338] agilent-dmm: Sending 'FETC? @2'.
sr: [01:18.526383] serial: Wrote 11/11 bytes.
...

knarfS, 2020-12-07:
That looks like a bug in the sigrok driver for the U1252A DMM. In the [driver](https://sigrok.org/gitweb/?p=libsigrok.git;a=blob;f=src/hardware/agilent-dmm/api.c), the U1252A is configured to have 3 channels, but something is wrong when the 2nd channel is selected by the driver (`FETC? @2`).

Unfortunately I couldn't find a protocol documentation (beside the [documentation from the sigrok wiki](https://sigrok.org/wiki/Agilent_U12xxx_series)) and therefore can't say anything about what exactly is wrong.

Maybe you can debug that further and submit a fix to sigrok?

ekute-dev, 2020-12-07:
You right,

I have to investigate further, it seems the 3 channels are due to this commit : https://github.com/sigrokproject/libsigrok/commit/d822726dbffea1fabe47bcea3e434eadcfa29b31#diff-fac17aaf75e51c33df1b0cf3405e2e9b94476fe59b83a0c7473aae2df5eb0cc5

closed 2020-12-07 by ekute-dev