knarfS/smuview #26: Fails to compile with Python 3.9 installed

peterska, 2021-01-04:
**Smuview compilation fails when python 3.9 is installed**
Compiling Smuview on a system with Python 3.9 produces a deprecation warning about PyEval_InitThreads() which in turn causes compilation to abort since warnings are being treated as errors

**To Reproduce**
1. mkdir buid
2. cd build
3. cmake ../
4. make

** To fix**
Python versions prior to 3.6 require the call to PyEval_InitThreads so some intelligence would need to be added to cmake.
edit smuview/external/pybind11/include/pybind11/detail/internals.h and comment out the call to PyEval_InitThreads() on line 275

**Screenshots or log**
```[ 18%] Building CXX object CMakeFiles/smuview.dir/smuview_autogen/mocs_compilation.cpp.o
In file included from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/cast.h:16,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/attr.h:13,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/pybind11.h:44,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/3MH5VUM4LI/../../../src/python/pystreamredirect.hpp:27,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/3MH5VUM4LI/moc_pystreamredirect.cpp:10,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/mocs_compilation.cpp:36:
/data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/detail/internals.h: In function ‘pybind11::detail::internals& pybind11::detail::get_internals()’:
/data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/detail/internals.h:276:28: error: ‘void PyEval_InitThreads()’ is deprecated [-Werror=deprecated-declarations]
  276 |         PyEval_InitThreads();
      |                            ^
In file included from /usr/include/python3.9/Python.h:145,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/detail/common.h:112,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/pytypes.h:12,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/cast.h:13,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/attr.h:13,
                 from /data/src/electronics/sigrok/smuview/external/pybind11/include/pybind11/pybind11.h:44,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/3MH5VUM4LI/../../../src/python/pystreamredirect.hpp:27,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/3MH5VUM4LI/moc_pystreamredirect.cpp:10,
                 from /data/src/electronics/sigrok/smuview/build/smuview_autogen/mocs_compilation.cpp:36:
/usr/include/python3.9/ceval.h:130:37: note: declared here
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/smuview.dir/build.make:133: CMakeFiles/smuview.dir/smuview_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:185: CMakeFiles/smuview.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

**Enviroment (Fedora 33):**
 - SmuView version / HEAD /master
 - OS: [Fedora 33.]

knarfS, 2021-01-04:
Thanks for letting me know.
This was fixed in https://github.com/pybind/pybind11/commit/ae2ee2a4a51914cf78deb5c253a3b03cbb53ce72. I'm in the process of updating the included pybind11 right now.

knarfS, 2021-01-06:
Fixed with commit https://github.com/knarfS/smuview/commit/9f0812925fa90997e71c703c9330f1c172211359

closed 2021-01-06 by knarfS