knarfS/smuview #27: Hardware support request Maynuo M9812

sympoz, 2021-01-05:
Is there any chance for support of the Maynuo M9812?

https://sigrok.org/wiki/Maynuo_M9812

knarfS, 2021-01-05:
The Maynuo M9812 is already supported by libsigrok and can be connected to SmuView, but the driver needs some small modifications (adding a mutex to make the driver thread safe) to work correctly with SmuView. There are also some other enhancements that would be beneficial for the combination with SmuView.

I'd be happy when someone would step up to make this driver work with SmuView. I can't do this without having that device next to me.

knarfS, 2021-01-05:
For reference: Here I've already added the mutex for an other modbus device: https://github.com/sigrokproject/libsigrok/commit/7c0891b0b82049b4cc3c0dd70b8483d037dab36f

mmuman, 2025-08-24:
I just ordered an M9710 second-hand, I'll have to tackle that.

closed 2026-08-17 by knarfS