knarfS/smuview #29: smuview doesn't recognize GW-INSTEK GDM-397

mickpf, 2021-04-05:
Hello Frank,

(BTW: Do you speak German? My English isn't well...)

Few days ago I bought a GW-INSTEK GDM-397 with a serial interface and compiled smuview 0.0.5 under ubuntu 20.04.
I've connected the multimeter to my PC via an USB-to-serial converter (PL2303), which is supported by linux and I use sometimes. udev rules a installed fine and work.
But smuview doesn't recognize it!.
I select the driver "GW Instek GDM-397 (gwinstek-gdm-397)", select the correct serial port "/dev/ttyUSB0", press the button "Scan for devices", few seconds long the software tries to scan and the list "Step 4" remains empty.

Do you have any idea, what can I do to check the reason?

Thanks in advance,
Michael

mickpf, 2021-04-05:
Sorry to wake you up! :-)

I found the right text in the manual: the device was in the sleep mode. I had to wake it up by pressing and holding the button "REL" on the device.
Now it works...

Thanks...

closed 2021-04-05 by mickpf

knarfS, 2021-04-05:
Great to hear, that it is working!