knarfS/smuview #32: CheckSigrokFeatures.cmake fails if PKGDEPS_STATIC_LIBRARY_DIRS or PKGDEPS_LIBRARY_DIRS is empty

peterska, 2021-06-15:
**Describe the bug**
When running configure on Fedora 34 it fails with your copy of sigrok is too old.
I traced this to the -L linker flag consuming the -lsigrokcxx flag as there are no linker search paths passed.

**Enviroment (please complete the following information):**
 - SmuView beb6cf4ca4f6da8b04299c744010ba30d5f1af17
 - OS: Fedora 34

Here is a patch that fixes the issue
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a7c1994..c1e32fc 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -98,9 +98,15 @@ pkg_check_modules(PKGDEPS REQUIRED ${PKGDEPS})
 
 include(CheckSigrokFeatures)
 if(STATIC_PKGDEPS_LIBS)
-       check_libsigrok_features("${PKGDEPS_STATIC_INCLUDE_DIRS}" "-L${PKGDEPS_STATIC_LIBRARY_DIRS};${PKGDEPS_STATIC_LDFLAGS}")
+       if(PKGDEPS_STATIC_LIBRARY_DIRS)
+               set(PKGDEPS_STATIC_LIBRARY_DIRS_SEARCH_FLAG "-L${PKGDEPS_STATIC_LIBRARY_DIRS}")
+       endif()
+       check_libsigrok_features("${PKGDEPS_STATIC_INCLUDE_DIRS}" "${PKGDEPS_STATIC_LIBRARY_DIRS_SEARCH_FLAG};${PKGDEPS_STATIC_LDFLAGS}")
 else()
-       check_libsigrok_features("${PKGDEPS_INCLUDE_DIRS}" "-L${PKGDEPS_LIBRARY_DIRS};${PKGDEPS_LIBRARIES}")
+       if(PKGDEPS_LIBRARY_DIRS)
+               set(PKGDEPS_LIBRARY_DIRS_SEARCH_FLAG "-L${PKGDEPS_LIBRARY_DIRS}")
+       endif()
+       check_libsigrok_features("${PKGDEPS_INCLUDE_DIRS}" "${PKGDEPS_LIBRARY_DIRS_SEARCH_FLAG};${PKGDEPS_LIBRARIES}")
 endif()
 
 set(CMAKE_INCLUDE_CURRENT_DIR ON)
```

knarfS, 2021-06-22:
Thanks for your feedback!
Coincidentally a similar problem occurred about three weeks ago when working on the CI for SmuView. I solved the issue by using a imported target for the pkg config stuff, which as a side effect also makes things nicer.

I cherry-picked the relevant commit (https://github.com/knarfS/smuview/commit/41a86c00887cfc1b23802e6559d4958d57fd424d) into master and I would be happy to hear from you, if this also solves the build problem on Fedora.

peterska, 2021-06-22:
Tried your patch on Fedora 34 and can confirm it now builds correctly.

closed 2021-06-22 by peterska