knarfS/smuview #33: Add support for RDTech RD6006

emeyedeejay, 2021-06-25:
Hi there ... is it possible to have support for the RDTech RD6006 (W) power supply added to SV, please?
The software that comes with the PSU is awful!

How can I help?

TIA
MJ

emeyedeejay, 2021-06-27:
Just to add, It seems the RD6006 is supported by sigrok. I'm not entirely sure what this means to be honest 😆

knarfS, 2021-06-27:
Yes, the RD6006 is supported in libsigrok since end of April (but no idea about the WiFi version). The last stable release of SmuView is based on an older version of libsigrok, so no RD6006 support in SmuView yet.

If you are on Linux, you can build SmuView by yourself, but if you aren't keen building it yourself or if you are on Mac/Win then I have good news for you: 
Right now I'm finalizing continuous builds for SmuView, so in the near future, up-to-date versions will always be available and you don't have to wait month until I'm in the mood of releasing a new version :D

emeyedeejay, 2021-06-28:
Woo hoo! Thanks! I understood most of that lol.

Where will the availability be announced etc.

Thanks!
MJ

knarfS, 2021-07-03:
Continuous Builds ("always up-to-date") are now available here: https://github.com/knarfS/smuview/releases/tag/continuous

Please let my know, how the RD6006 works! ;)

closed 2021-07-18 by knarfS