knarfS/smuview #35: Itech IT8511A+ segmentation fault

spike77453, 2021-08-01:
**Describe the bug**
Adding a new IT8511A+ device results in segmentation fault

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Add device'
2. Pick ITECH IT8500 driver
3. Select 'Serial port'
4. Set baud rate to 115200
5. Click on 'Scan for devices using driver above'
6. Select 'ITECH 8511A with 3 channels'
7. Click on 'OK'

**Expected behavior**
A new Itech 8511A+ device in the device tree

**Screenshots or log**
[image]

Clicking 'OK' in the screenshot above crashes smuview

**Enviroment (please complete the following information):**
 - SmuView version: 7f2a9c7f5fe11c804c86a6f98d0b527095699efd
 - self compiled,  gcc version 11.2.1
 - OS: Fedora 34

**Additional context**

The libsigrok version `0.5.2-7.20210801gitb96051a` in the backtrace below is compiled from git commit https://github.com/sigrokproject/libsigrok/commit/b96051a507100543bef235365b2ce68942d5f0d8 (can be found here: https://copr.fedorainfracloud.org/coprs/spike/sigrok/build/2356228/).

```
Thread 12 "smuview" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc53c1640 (LWP 202186)]
0x0000000000548814 in std::__shared_ptr<sv::data::BaseSignal, (__gnu_cxx::_Lock_policy)2>::operator= (this=0x1dbb958) at /usr/include/c++/11/bits/shared_ptr_base.h:1153
1153          __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
Missing separate debuginfos, use: dnf debuginfo-install bzip2-libs-1.0.8-6.fc34.x86_64 dbus-libs-1.12.20-3.fc34.x86_64 dbusmenu-qt5-0.9.3-0.26.20160218.fc34.x86_64 fontconfig-2.13.94-2.fc34.x86_64 freetype-2.10.4-3.fc34.x86_64 glib2-2.68.2-1.fc34.x86_64 glibc-2.33-20.fc34.x86_64 glibmm24-2.66.1-1.fc34.x86_64 graphite2-1.3.14-7.fc34.x86_64 harfbuzz-2.7.4-3.fc34.x86_64 jasper-libs-2.0.32-1.fc34.x86_64 jbigkit-libs-2.1-21.fc34.x86_64 jpegxl-libs-0.3.7-3.fc34.x86_64 keyutils-libs-1.6.1-2.fc34.x86_64 kf5-karchive-5.83.0-1.fc34.x86_64 kf5-kbookmarks-5.83.0-1.fc34.x86_64 kf5-kcodecs-5.83.0-1.fc34.x86_64 kf5-kcompletion-5.83.0-1.fc34.x86_64 kf5-kconfig-core-5.83.0-1.fc34.x86_64 kf5-kconfig-gui-5.83.0-1.fc34.x86_64 kf5-kconfigwidgets-5.83.0-1.fc34.x86_64 kf5-kcrash-5.83.0-1.fc34.x86_64 kf5-kdbusaddons-5.83.0-1.fc34.x86_64 kf5-kglobalaccel-libs-5.83.0-1.fc34.x86_64 kf5-ki18n-5.83.0-1.fc34.x86_64 kf5-kiconthemes-5.83.0-1.fc34.x86_64 kf5-kimageformats-5.83.0-1.fc34.x86_64 kf5-kio-core-libs-5.83.0-2.fc34.x86_64 kf5-kio-file-widgets-5.83.0-2.fc34.x86_64 kf5-kio-gui-5.83.0-2.fc34.x86_64 kf5-kitemviews-5.83.0-1.fc34.x86_64 kf5-kjobwidgets-5.83.0-1.fc34.x86_64 kf5-knotifications-5.83.0-1.fc34.x86_64 kf5-kservice-5.83.0-1.fc34.x86_64 kf5-kwayland-5.83.0-1.fc34.x86_64 kf5-kwidgetsaddons-5.83.0-1.fc34.x86_64 kf5-kwindowsystem-5.83.0-1.fc34.x86_64 kf5-solid-5.83.0-1.fc34.x86_64 krb5-libs-1.19.1-14.fc34.x86_64 lcms2-2.12-1.fc34.x86_64 libICE-1.0.10-6.fc34.x86_64 libX11-1.7.0-3.fc34.x86_64 libX11-xcb-1.7.0-3.fc34.x86_64 libXau-1.0.9-6.fc34.x86_64 libXcursor-1.2.0-5.fc34.x86_64 libXext-1.3.4-6.fc34.x86_64 libXfixes-6.0.0-1.fc34.x86_64 libXtst-1.2.3-14.fc34.x86_64 libacl-2.3.1-1.fc34.x86_64 libaom-3.1.1-1.fc34.x86_64 libattr-2.5.1-1.fc34.x86_64 libbrotli-1.0.9-4.fc34.x86_64 libcanberra-0.30-24.fc34.x86_64 libcap-2.48-2.fc34.x86_64 libcom_err-1.45.6-5.fc34.x86_64 libdav1d-0.9.0-1.fc34.x86_64 libdrm-2.4.107-1.fc34.x86_64 libffi-3.1-28.fc34.x86_64 libftdi-1.4-7.fc34.x86_64 libgcc-11.2.1-1.fc34.x86_64 libgcrypt-1.9.3-3.fc34.x86_64 libglvnd-1.3.3-1.fc34.x86_64 libglvnd-glx-1.3.3-1.fc34.x86_64 libgpg-error-1.42-1.fc34.x86_64 libicu-67.1-6.fc34.x86_64 libieee1284-0.2.11-35.fc34.x86_64 libmng-2.0.3-13.fc34.x86_64 libmount-2.36.2-1.fc34.x86_64 libogg-1.3.4-4.fc34.x86_64 libpciaccess-0.16-4.fc34.x86_64 libpng-1.6.37-10.fc34.x86_64 libserialport-0.1.1-11.fc34.x86_64 libsigc++20-2.10.7-1.fc34.x86_64 libsigrok-0.5.2-7.20210801gitb96051a.fc34.x86_64 libsigrok-cxx-0.5.2-7.20210801gitb96051a.fc34.x86_64 libstdc++-11.2.1-1.fc34.x86_64 libtdb-1.4.3-6.fc34.x86_64 libtirpc-1.3.2-0.fc34.x86_64 libtool-ltdl-2.4.6-40.fc34.x86_64 libusbx-1.0.24-2.fc34.x86_64 libuuid-2.36.2-1.fc34.x86_64 libvorbis-1.3.7-3.fc34.x86_64 libwayland-client-1.19.0-1.fc34.x86_64 libwayland-cursor-1.19.0-1.fc34.x86_64 libwebp-1.2.0-1.fc34.x86_64 libxcb-1.13.1-7.fc34.x86_64 libxkbcommon-x11-1.3.0-1.fc34.x86_64 libxml2-2.9.12-4.fc34.x86_64 libxshmfence-1.3-8.fc34.x86_64 libzip-1.7.3-4.fc34.x86_64 libzstd-1.5.0-1.fc34.x86_64 lz4-libs-1.9.3-2.fc34.x86_64 mesa-libGL-21.1.5-1.fc34.x86_64 mesa-libglapi-21.1.5-1.fc34.x86_64 openssl-libs-1.1.1k-1.fc34.x86_64 pcre-8.44-3.fc34.1.x86_64 pcre2-10.36-4.fc34.x86_64 pcre2-utf16-10.36-4.fc34.x86_64 python3-libs-3.9.6-2.fc34.x86_64 qt5-qtbase-5.15.2-16.fc34.x86_64 qt5-qtbase-gui-5.15.2-16.fc34.x86_64 qt5-qtquickcontrols2-5.15.2-3.fc34.x86_64 qt5-qtsvg-5.15.2-4.fc34.x86_64 qt5-qtwayland-5.15.2-7.fc34.x86_64 qwt-qt5-6.1.5-4.fc34.x86_64 rav1e-libs-0.4.1-2.fc34.x86_64 sssd-client-2.5.2-1.fc34.x86_64 svt-av1-libs-0.8.6-4.fc34.x86_64 systemd-libs-248.6-1.fc34.x86_64
(gdb) backtrace
#0  0x0000000000548814 in std::__shared_ptr<sv::data::BaseSignal, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<sv::data::BaseSignal, (__gnu_cxx::_Lock_policy)2> const&)
    (this=0x1dbb958) at /usr/include/c++/11/bits/shared_ptr_base.h:1153
#1  0x0000000000548863 in std::shared_ptr<sv::data::BaseSignal>::operator=(std::shared_ptr<sv::data::BaseSignal> const&) (this=0x1dbb958) at /usr/include/c++/11/bits/shared_ptr.h:359
#2  0x000000000055811f in sv::channels::HardwareChannel::push_interleaved_samples(float const*, unsigned long, unsigned long, double, unsigned long, std::shared_ptr<sigrok::Analog>)
    (this=
    0x1dbb8b0, data=0x7fffb40046d0, sample_count=1, stride=1, timestamp=1627857685.4949999, samplerate=10, sr_analog=std::shared_ptr<sigrok::Analog> (use count 3, weak count 1) = {...}) at /home/spike/workspace/smuview/smuview-7f2a9c7f5fe11c804c86a6f98d0b527095699efd/src/channels/hardwarechannel.cpp:100
#3  0x00000000006d7f7c in sv::devices::HardwareDevice::feed_in_analog(std::shared_ptr<sigrok::Analog>)
    (this=0x1b8b150, sr_analog=std::shared_ptr<sigrok::Analog> (use count 3, weak count 1) = {...})
    at /home/spike/workspace/smuview/smuview-7f2a9c7f5fe11c804c86a6f98d0b527095699efd/src/devices/hardwaredevice.cpp:296
#4  0x000000000068bd23 in sv::devices::BaseDevice::data_feed_in(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)
    (this=0x1b8b150, sr_device=std::shared_ptr<sigrok::Device> (use count 10, weak count 1) = {...}, sr_packet=std::shared_ptr<sigrok::Packet> (use count 3, weak count 1) = {...})
    at /home/spike/workspace/smuview/smuview-7f2a9c7f5fe11c804c86a6f98d0b527095699efd/src/devices/basedevice.cpp:478
#5  0x000000000068b93d in operator()(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>) const
    (__closure=0x1dc9790, sr_device=std::shared_ptr<sigrok::Device> (use count 10, weak count 1) = {...}, sr_packet=std::shared_ptr<sigrok::Packet> (use count 3, weak count 1) = {...})
    at /home/spike/workspace/smuview/smuview-7f2a9c7f5fe11c804c86a6f98d0b527095699efd/src/devices/basedevice.cpp:418
#6  0x000000000068c641 in std::__invoke_impl<void, sv::devices::BaseDevice::init_acquisition()::<lambda(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)>&, std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet> >(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/11/bits/invoke.h:61
#7  0x000000000068c4d8 in std::__invoke_r<void, sv::devices::BaseDevice::init_acquisition()::<lambda(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)>&, std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet> >(struct {...} &) (__fn=...) at /usr/include/c++/11/bits/invoke.h:154
#8  0x000000000068c393 in std::_Function_handler<void(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>), sv::devices::BaseDevice::init_acquisition()::<lambda(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)> >::_M_invoke(const std::_Any_data &, std::shared_ptr<sigrok::Device> &&, std::shared_ptr<sigrok::Packet> &&)
    (__functor=..., __args#0=..., __args#1=...) at /usr/include/c++/11/bits/std_function.h:291
#9  0x00007ffff7da5251 in sigrok::datafeed_callback(sr_dev_inst const*, sr_datafeed_packet const*, void*) () at /lib64/libsigrokcxx.so.4
#10 0x00007ffff7c63508 in sr_session_send () at /lib64/libsigrok.so.4
#11 0x00007ffff7d0384d in itech_it8500_channel_send_value.constprop () at /lib64/libsigrok.so.4
[... cut ...]
#23 0x00000000006a434a in std::thread::_Invoker<std::tuple<void (sv::devices::BaseDevice::*)(), sv::devices::BaseDevice*> >::operator()() (this=0x1dc9b68) at /usr/include/c++/11/bits/std_thread.h:260
#24 0x00000000006a42ae in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (sv::devices::BaseDevice::*)(), sv::devices::BaseDevice*> > >::_M_run() (this=0x1dc9b60) at /usr/include/c++/11/bits/std_thread.h:211
#25 0x00007ffff6284c84 in execute_native_thread_routine () at /lib64/libstdc++.so.6
#26 0x00007ffff7dcc299 in start_thread () at /lib64/libpthread.so.0
#27 0x00007ffff5f7d353 in clone () at /lib64/libc.so.6
```

spike77453, 2021-08-02:
With the pre-built x86_64 AppImage from https://github.com/knarfS/smuview/releases/tag/continuous the trace looks a bit different:

```
Thread 12 "smuview" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd2304640 (LWP 373898)]
0x000055555585f3a9 in sv::channels::HardwareChannel::push_interleaved_samples(float const*, unsigned long, unsigned long, double, unsigned long, std::shared_ptr<sigrok::Analog>) ()
Missing separate debuginfos, use: dnf debuginfo-install bzip2-libs-1.0.8-6.fc34.x86_64 expat-2.4.1-1.fc34.x86_64 fontconfig-2.13.94-2.fc34.x86_64 freetype-2.10.4-3.fc34.x86_64 glib2-2.68.2-1.fc34.x86_64 glibc-2.33-20.fc34.x86_64 graphite2-1.3.14-7.fc34.x86_64 harfbuzz-2.7.4-3.fc34.x86_64 libICE-1.0.10-6.fc34.x86_64 libX11-1.7.0-3.fc34.x86_64 libXau-1.0.9-6.fc34.x86_64 libXcursor-1.2.0-5.fc34.x86_64 libXext-1.3.4-6.fc34.x86_64 libXfixes-6.0.0-1.fc34.x86_64 libXxf86vm-1.1.4-16.fc34.x86_64 libbrotli-1.0.9-4.fc34.x86_64 libdrm-2.4.107-1.fc34.x86_64 libffi-3.1-28.fc34.x86_64 libgcc-11.2.1-1.fc34.x86_64 libglvnd-1.3.3-1.fc34.x86_64 libglvnd-glx-1.3.3-1.fc34.x86_64 libgpg-error-1.42-1.fc34.x86_64 libmount-2.36.2-1.fc34.x86_64 libstdc++-11.2.1-1.fc34.x86_64 libusbx-1.0.24-2.fc34.x86_64 libuuid-2.36.2-1.fc34.x86_64 libxcb-1.13.1-7.fc34.x86_64 libxml2-2.9.12-4.fc34.x86_64 libxshmfence-1.3-8.fc34.x86_64 mesa-dri-drivers-21.1.5-1.fc34.x86_64 mesa-libGL-21.1.5-1.fc34.x86_64 mesa-libglapi-21.1.5-1.fc34.x86_64 pcre-8.44-3.fc34.1.x86_64 pcre2-10.36-4.fc34.x86_64 sssd-client-2.5.2-1.fc34.x86_64 zlib-1.2.11-26.fc34.x86_64
(gdb) backtrace
#0  0x000055555585f3a9 in sv::channels::HardwareChannel::push_interleaved_samples(float const*, unsigned long, unsigned long, double, unsigned long, std::shared_ptr<sigrok::Analog>) ()
#1  0x0000555555882597 in sv::devices::HardwareDevice::feed_in_analog(std::shared_ptr<sigrok::Analog>) ()
#2  0x0000555555874e31 in sv::devices::BaseDevice::data_feed_in(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>) ()
#3  0x0000555555874fa7 in std::_Function_handler<void (std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>), sv::devices::BaseDevice::init_acquisition()::{lambda(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<sigrok::Device>&&, std::shared_ptr<sigrok::Packet>&&) ()
#4  0x00007ffff7a01b95 in  () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrokcxx.so.4
#5  0x00007ffff76a764f in  () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrok.so.4
#6  0x00007ffff770fcf5 in  () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrok.so.4
#7  0x00007ffff770fe03 in  () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrok.so.4
#8  0x00007ffff76a69c0 in  () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrok.so.4
#9  0x00007ffff72eb4cf in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#10 0x00007ffff733f4e8 in g_main_context_iterate.constprop () at /lib64/libglib-2.0.so.0
#11 0x00007ffff72eaa93 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#12 0x00007ffff76a7371 in sr_session_run () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrok.so.4
#13 0x00007ffff79f513d in sigrok::Session::run() () at /tmp/.mount_SmuVieXMxKEe/usr/bin/../lib/libsigrokcxx.so.4
#14 0x000055555587311b in sv::devices::BaseDevice::aquisition_thread_proc() ()
#15 0x00007ffff5b25c84 in execute_native_thread_routine () at /lib64/libstdc++.so.6
#16 0x00007ffff7c24299 in start_thread () at /lib64/libpthread.so.0
#17 0x00007ffff581c353 in clone () at /lib64/libc.so.6
```

spike77453, 2021-08-18:
Compiling with `-Wp,-D_GLIBCXX_ASSERTIONS` leads to:

```
HardwareChannel::push_sample_sr_analog():  "P1"  - No signal found:  "P1 [W]"
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::shared_ptr<sv::data::BaseSignal>; _Alloc = std::allocator<std::shared_ptr<sv::data::BaseSignal> >; std::vector<_Tp, _Alloc>::reference = std::shared_ptr<sv::data::BaseSignal>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Thread 12 "smuview" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcd22d640 (LWP 525438)]
0x00007ffff5f7d2a2 in raise () from /lib64/libc.so.6
(gdb) backtrace
#0  0x00007ffff5f7d2a2 in raise () at /lib64/libc.so.6
#1  0x00007ffff5f668a4 in abort () at /lib64/libc.so.6
#2  0x00005555559eaca8 in  ()
#3  0x0000555555a4df29 in sv::devices::HardwareDevice::feed_in_analog(std::shared_ptr<sigrok::Analog>) ()
#4  0x0000555555a400eb in std::_Function_handler<void (std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>), sv::devices::BaseDevice::init_acquisition()::{lambda(std::shared_ptr<sigrok::Device>, std::shared_ptr<sigrok::Packet>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<sigrok::Device>&&, std::shared_ptr<sigrok::Packet>&&) ()
#5  0x00007ffff7e04251 in sigrok::datafeed_callback(sr_dev_inst const*, sr_datafeed_packet const*, void*) () at /lib64/libsigrokcxx.so.4
#6  0x00007ffff7cc2508 in sr_session_send () at /lib64/libsigrok.so.4
#7  0x00007ffff7d6294d in itech_it8500_channel_send_value.constprop () at /lib64/libsigrok.so.4
#8  0x00007ffff7d1ff3d in itech_it8500_receive_data () at /lib64/libsigrok.so.4
#9  0x00007ffff7cc1c85 in fd_source_dispatch () at /lib64/libsigrok.so.4
#10 0x00007ffff7b0e4cf in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#11 0x00007ffff7b624e8 in g_main_context_iterate.constprop () at /lib64/libglib-2.0.so.0
#12 0x00007ffff7b0da93 in g_main_loop_run () at /lib64/libglib-2.0.so.0
#13 0x00007ffff7cc21d5 in sr_session_run () at /lib64/libsigrok.so.4
#14 0x00007ffff7e026d4 in sigrok::Session::run() () at /lib64/libsigrokcxx.so.4
#15 0x0000555555a3fb4e in sv::devices::BaseDevice::aquisition_thread_proc() ()
#16 0x00007ffff6349c84 in execute_native_thread_routine () at /lib64/libstdc++.so.6
#17 0x00007ffff7e2b299 in start_thread () at /lib64/libpthread.so.0
#18 0x00007ffff6040353 in clone () at /lib64/libc.so.6
```

knarfS, 2021-08-20:
Could you please start SmuView with full log output and attach the whole output here?

This is for connecting the load with the (factory) default serial settings (9600/8n1). Maybe you have to adapt the TTY:
```
./smuview -d itech-it8500:conn=/dev/ttyUSB0 -l 5
```

When looking into the sigrok device driver, 38400 should be the maximum baud rate (not sure, I don't own that device):
```
./smuview -d itech-it8500:conn=/dev/ttyUSB0:serialcomm=38400/8n1 -l 5
```

spike77453, 2021-08-21:
Just for the record, factory default settings on my unit is 115200 8N1. I've set it to 38400 though:

```
$ smuview -d itech-it8500:conn=/dev/ttyUSB0:serialcomm=38400/8n1 -l 5
sr: [00:00.000000] log: libsigrok loglevel set to 5.
sr: [00:00.027333] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.028219] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.048299] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.048493] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.048799] hwdriver: Scan found 0 devices (baylibre-acme).
sr: [00:00.048915] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.049008] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.049084] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.049099] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.049106] hwdriver: Scan found 0 devices (brymen-bm52x).
sr: [00:00.049263] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.049279] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.049286] hwdriver: Scan found 0 devices (brymen-bm82x).
sr: [00:00.049388] hwdriver: Scan found 0 devices (brymen-bm85x).
sr: [00:00.049506] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.049524] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.049534] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:00.049625] hwdriver: Scan found 0 devices (cem-dt-885x).
sr: [00:00.049732] hwdriver: Scan found 0 devices (center-309).
sr: [00:00.049841] hwdriver: Scan found 0 devices (colead-slm).
sr: [00:00.049972] hwdriver: Scan found 0 devices (conrad-digi-35-cpu).
sr: [00:00.050072] demo: Generating square pattern.
sr: [00:00.050090] demo: Generating sine pattern.
sr: [00:00.050135] demo: Generating triangle pattern.
sr: [00:00.050209] demo: Generating sawtooth pattern.
sr: [00:00.050238] hwdriver: Scan found 1 devices (demo).
sr: [00:00.050357] hwdriver: Scan found 0 devices (deree-de5000).
sr: [00:00.050510] hwdriver: Scan found 0 devices (digitek-dt4000zc).
sr: [00:00.053953] hwdriver: Scan found 0 devices (eevblog-121gw).
sr: [00:00.056340] serial: No serial device specified.
sr: [00:00.062737] hwdriver: Scan found 0 devices (fluke-45).
sr: [00:00.114929] hwdriver: Scan found 0 devices (fluke-dmm).
sr: [00:00.115643] hwdriver: Scan found 0 devices (gmc-mh-1x-2x-rs232).
sr: [00:00.116282] hwdriver: Scan found 0 devices (gmc-mh-2x-bd232).
sr: [00:00.117007] hwdriver: Scan found 0 devices (gwinstek-gdm-397).
sr: [00:00.117579] hwdriver: Scan found 0 devices (gwinstek-gpd).
sr: [00:00.118181] serial: No serial device specified.
sr: [00:00.118196] hwdriver: Scan found 0 devices (hp-3457a).
sr: [00:00.118688] serial: No serial device specified.
sr: [00:00.118708] hwdriver: Scan found 0 devices (hp-59306a).
sr: [00:00.119084] serial: No serial device specified.
sr: [00:00.119097] hwdriver: Scan found 0 devices (hpib-pps).
sr: [00:00.119448] hwdriver: Scan found 0 devices (iso-tech-idm103n).
sr: [00:00.120047] hwdriver: Scan found 0 devices (kecheng-kc-330b).
sr: [00:00.120472] hwdriver: Scan found 0 devices (kern-ew-6200-2nm).
sr: [00:00.120917] hwdriver: Scan found 0 devices (korad-kaxxxxp).
sr: [00:00.121279] hwdriver: Scan found 0 devices (lascar-el-usb).
sr: [00:00.121687] hwdriver: Scan found 0 devices (manson-hcs-3xxx).
sr: [00:00.122272] hwdriver: Scan found 0 devices (mastech-mas345).
sr: [00:00.122668] hwdriver: Scan found 0 devices (mastech-ms2115b).
sr: [00:00.123114] hwdriver: Scan found 0 devices (mastech-ms5308).
sr: [00:00.123757] hwdriver: Scan found 0 devices (mastech-ms6514).
sr: [00:00.124220] hwdriver: Scan found 0 devices (mastech-ms8250b).
sr: [00:00.124840] hwdriver: Scan found 0 devices (mastech-ms8250d).
sr: [00:00.125315] hwdriver: Scan found 0 devices (maynuo-m97).
sr: [00:00.126197] hwdriver: Scan found 0 devices (meterman-38xr).
sr: [00:00.126816] hwdriver: Scan found 0 devices (metex-m3640d).
sr: [00:00.127202] hwdriver: Scan found 0 devices (metex-m3860m).
sr: [00:00.127668] hwdriver: Scan found 0 devices (metex-m4650cr).
sr: [00:00.128109] hwdriver: Scan found 0 devices (metex-me21).
sr: [00:00.128447] hwdriver: Scan found 0 devices (metex-me31).
sr: [00:00.128837] hwdriver: Scan found 0 devices (metrix-mx56c).
sr: [00:00.129306] hwdriver: Scan found 0 devices (mic-98581).
sr: [00:00.129660] hwdriver: Scan found 0 devices (mic-98583).
sr: [00:00.129979] serial: No serial device specified.
[... 209 lines cut ...]
sr: [00:00.480775] serial: Read 26/26 bytes.
sr: [00:00.480806] itech-it8500: itech_it8500_send_cmd: Response packet received: cmd=5f
sr: [00:00.480823] itech-it8500: Load status: V=0,0000, I=0,0000, P=0,000, State=OFF, Mode=CC (op=0x15, demand=0x0040)
sr: [00:00.480833] itech-it8500: itech_it8500_send_cmd: Sending command: 5f
sr: [00:00.480853] hwdriver: sr_config_get(): key 30043 (regulation) sdi 0x55f1006ccc30 cg 1 -> 'CC'
sr: [00:00.481159] serial: Wrote 26/26 bytes.
sr: [00:00.495667] serial: Read 26/26 bytes.
sr: [00:00.495686] itech-it8500: itech_it8500_send_cmd: Response packet received: cmd=5f
sr: [00:00.495698] sr: [00:00.495702] itech-it8500: itech_it8500_send_cmd: Sending command: 5f
itech-it8500: Load status: V=0,0000, I=0,0000, P=0,000, State=OFF, Mode=CC (op=0x15, demand=0x0040)
sr: [00:00.495723] session: bus: Received SR_DF_FRAME_BEGIN packet.
[... cut ...]
HardwareChannel::push_sample_sr_analog():  "P1"  - No signal found:  "P1 [W]"
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::shared_ptr<sv::data::BaseSignal>; _Alloc = std::allocator<std::shared_ptr<sv::data::BaseSignal> >; std::vector<_Tp, _Alloc>::reference = std::shared_ptr<sv::data::BaseSignal>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

knarfS, 2021-08-22:
Fixed :)
The cause of this issue was, that SmuView couldn't handle the unexpected AC flag for the power channel.

Now signals can be added to "fixed" channels: https://github.com/knarfS/smuview/commit/ca7951b2cc9db614c152123171960b883aaa9bc5
And the AC flag was removed in the sigrok driver: https://github.com/sigrokproject/libsigrok/commit/24be1e05d5f1cbe84c69b0bd27d422ad4b13cd77

spike77453, 2021-08-23:
Thanks! Looks different now:

```
$ smuview -d itech-it8500:conn=/dev/ttyUSB0:serialcomm=38400/8n1 -l 5
sr: [00:00.000001] log: libsigrok loglevel set to 5.
sr: [00:00.026219] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.027137] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.044971] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.045297] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.045759] hwdriver: Scan found 0 devices (baylibre-acme).
sr: [00:00.045979] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.046208] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.046354] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.046390] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.046410] hwdriver: Scan found 0 devices (brymen-bm52x).
sr: [00:00.046667] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.046686] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.046694] hwdriver: Scan found 0 devices (brymen-bm82x).
sr: [00:00.046819] hwdriver: Scan found 0 devices (brymen-bm85x).
sr: [00:00.046988] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.047004] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.047013] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:00.047200] hwdriver: Scan found 0 devices (cem-dt-885x).
sr: [00:00.047356] hwdriver: Scan found 0 devices (center-309).
sr: [00:00.047483] hwdriver: Scan found 0 devices (colead-slm).
sr: [00:00.047605] hwdriver: Scan found 0 devices (conrad-digi-35-cpu).
sr: [00:00.047713] demo: Generating square pattern.
sr: [00:00.047738] demo: Generating sine pattern.
sr: [00:00.047793] demo: Generating triangle pattern.
sr: [00:00.047866] demo: Generating sawtooth pattern.
sr: [00:00.047902] hwdriver: Scan found 1 devices (demo).
sr: [00:00.048030] hwdriver: Scan found 0 devices (deree-de5000).
sr: [00:00.048157] hwdriver: Scan found 0 devices (digitek-dt4000zc).
sr: [00:00.048300] hwdriver: Scan found 0 devices (eevblog-121gw).
sr: [00:00.087568] serial: No serial device specified.
sr: [00:00.092351] hwdriver: Scan found 0 devices (fluke-45).
sr: [00:00.095193] hwdriver: Scan found 0 devices (fluke-dmm).
sr: [00:00.096888] hwdriver: Scan found 0 devices (gmc-mh-1x-2x-rs232).
sr: [00:00.097610] hwdriver: Scan found 0 devices (gmc-mh-2x-bd232).
sr: [00:00.098018] hwdriver: Scan found 0 devices (gwinstek-gdm-397).
sr: [00:00.098468] hwdriver: Scan found 0 devices (gwinstek-gpd).
sr: [00:00.099002] serial: No serial device specified.
sr: [00:00.099016] hwdriver: Scan found 0 devices (hp-3457a).
sr: [00:00.099397] serial: No serial device specified.
sr: [00:00.099408] hwdriver: Scan found 0 devices (hp-59306a).
sr: [00:00.099757] serial: No serial device specified.
sr: [00:00.099769] hwdriver: Scan found 0 devices (hpib-pps).
sr: [00:00.100223] hwdriver: Scan found 0 devices (iso-tech-idm103n).
sr: [00:00.100701] hwdriver: Scan found 0 devices (kecheng-kc-330b).
sr: [00:00.101056] hwdriver: Scan found 0 devices (kern-ew-6200-2nm).
sr: [00:00.101627] hwdriver: Scan found 0 devices (korad-kaxxxxp).
sr: [00:00.102025] hwdriver: Scan found 0 devices (lascar-el-usb).
sr: [00:00.102374] hwdriver: Scan found 0 devices (manson-hcs-3xxx).
sr: [00:00.102755] hwdriver: Scan found 0 devices (mastech-mas345).
sr: [00:00.103144] hwdriver: Scan found 0 devices (mastech-ms2115b).
sr: [00:00.103548] hwdriver: Scan found 0 devices (mastech-ms5308).
sr: [00:00.104022] hwdriver: Scan found 0 devices (mastech-ms6514).
sr: [00:00.104529] hwdriver: Scan found 0 devices (mastech-ms8250b).
sr: [00:00.104856] hwdriver: Scan found 0 devices (mastech-ms8250d).
sr: [00:00.105289] hwdriver: Scan found 0 devices (maynuo-m97).
sr: [00:00.105764] hwdriver: Scan found 0 devices (meterman-38xr).
sr: [00:00.106086] hwdriver: Scan found 0 devices (metex-m3640d).
sr: [00:00.106424] hwdriver: Scan found 0 devices (metex-m3860m).
sr: [00:00.106746] hwdriver: Scan found 0 devices (metex-m4650cr).
sr: [00:00.107094] hwdriver: Scan found 0 devices (metex-me21).
sr: [00:00.107420] hwdriver: Scan found 0 devices (metex-me31).
sr: [00:00.107758] hwdriver: Scan found 0 devices (metrix-mx56c).
sr: [00:00.108064] hwdriver: Scan found 0 devices (mic-98581).
sr: [00:00.108372] hwdriver: Scan found 0 devices (mic-98583).
sr: [00:00.108785] serial: No serial device specified.
[... 195 lines cut ...]
sr: [00:00.383545] session: Creating our own main context.
sr: [00:00.383577] session: Starting.
sr: [00:00.383582] hwdriver: itech-it8500: Starting acquisition.
sr: [00:00.383597] session: bus: Received SR_DF_HEADER packet.
Start aquisition for  "ITECH 8511A" ,  aquisition_start_timestamp_ =  "2021.08.23 02:02:47.820"
BaseDevice::close(): Trying to close device  "ITECH 8511A 1.52 4757510274 (/dev/ttyUSB0)"
sr: [00:00.440322] session: Stopping.
sr: [00:00.440346] hwdriver: itech-it8500: Stopping acquisition.
sr: [00:00.440354] session: bus: Received SR_DF_END packet.
sr: [00:00.440377] session: fd_source_finalize: key 0x55df1c9b8a00
sr: [00:00.440397] session: Stopped.
sr: [00:00.440457] device: itech-it8500: Closing device instance.
sr: [00:00.440470] itech-it8500: itech_it8500_send_cmd: Sending command: 20
sr: [00:00.440593] serial: Wrote 26/26 bytes.
sr: [00:00.455079] serial: Read 26/26 bytes.
sr: [00:00.455100] itech-it8500: itech_it8500_send_cmd: Response packet received: cmd=12
sr: [00:00.455109] serial: Closing serial port /dev/ttyUSB0.
BaseDevice::close(): Device closed  "ITECH 8511A 1.52 4757510274 (/dev/ttyUSB0)"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"
main() failed:  invalid argument
```

New builds can be found here: https://copr.fedorainfracloud.org/coprs/spike/sigrok/builds/
Should I open a new issue?

knarfS, 2021-08-28:
I've looked into this and I think the Itech driver (for whatever reason) returns a `SR_ERR_ARG` somewhere down the line from `api.c`, `config_get()`, which finally throws an exception in the libsigrok C++ bindings (from your log: `main() failed:  invalid argument`).

I've done some experimental changes in SmuView to catch this exception and output some error messages, that hopefully will help us to figure the root issue.

It would be great, if you would build my SmuView `itech`branch (https://github.com/knarfS/smuview/tree/itech) and then attach another log.

spike77453, 2021-09-10:
Sorry, only now got around to doing this:

```
$ smuview -d itech-it8500:conn=/dev/ttyUSB0:serialcomm=9600/8n1 -l 5
sr: [00:00.000000] log: libsigrok loglevel set to 5.
sr: [00:00.024680] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.025596] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.045420] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.045631] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.046004] hwdriver: Scan found 0 devices (baylibre-acme).
sr: [00:00.046133] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.046246] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.046343] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.046357] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.046365] hwdriver: Scan found 0 devices (brymen-bm52x).
sr: [00:00.046489] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.046502] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.046506] hwdriver: Scan found 0 devices (brymen-bm82x).
sr: [00:00.046724] hwdriver: Scan found 0 devices (brymen-bm85x).
sr: [00:00.046927] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.046945] serial-libsp: Error getting port from name hid/bu86x: (2) No such file or directory.
sr: [00:00.046958] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:00.047067] hwdriver: Scan found 0 devices (cem-dt-885x).
sr: [00:00.050703] hwdriver: Scan found 0 devices (center-309).
sr: [00:00.052580] hwdriver: Scan found 0 devices (colead-slm).
sr: [00:00.053086] hwdriver: Scan found 0 devices (conrad-digi-35-cpu).
sr: [00:00.085462] demo: Generating square pattern.
sr: [00:00.085493] demo: Generating sine pattern.
sr: [00:00.085542] demo: Generating triangle pattern.
sr: [00:00.085606] demo: Generating sawtooth pattern.
sr: [00:00.085647] hwdriver: Scan found 1 devices (demo).
sr: [00:00.086587] hwdriver: Scan found 0 devices (deree-de5000).
sr: [00:00.088594] hwdriver: Scan found 0 devices (digitek-dt4000zc).
sr: [00:00.089060] hwdriver: Scan found 0 devices (eevblog-121gw).
sr: [00:00.089451] serial: No serial device specified.
sr: [00:00.101504] hwdriver: Scan found 0 devices (fluke-45).
sr: [00:00.102539] hwdriver: Scan found 0 devices (fluke-dmm).
sr: [00:00.103152] hwdriver: Scan found 0 devices (gmc-mh-1x-2x-rs232).
sr: [00:00.103591] hwdriver: Scan found 0 devices (gmc-mh-2x-bd232).
sr: [00:00.103981] hwdriver: Scan found 0 devices (gwinstek-gdm-397).
sr: [00:00.104356] hwdriver: Scan found 0 devices (gwinstek-gpd).
sr: [00:00.104820] serial: No serial device specified.
sr: [00:00.105008] hwdriver: Scan found 0 devices (hp-3457a).
sr: [00:00.105433] serial: No serial device specified.
sr: [00:00.105444] hwdriver: Scan found 0 devices (hp-59306a).
sr: [00:00.105892] serial: No serial device specified.
sr: [00:00.105905] hwdriver: Scan found 0 devices (hpib-pps).
sr: [00:00.106417] hwdriver: Scan found 0 devices (iso-tech-idm103n).
sr: [00:00.106969] hwdriver: Scan found 0 devices (kecheng-kc-330b).
sr: [00:00.107389] hwdriver: Scan found 0 devices (kern-ew-6200-2nm).
sr: [00:00.107995] hwdriver: Scan found 0 devices (korad-kaxxxxp).
sr: [00:00.108367] hwdriver: Scan found 0 devices (lascar-el-usb).
sr: [00:00.108739] hwdriver: Scan found 0 devices (manson-hcs-3xxx).
sr: [00:00.109148] hwdriver: Scan found 0 devices (mastech-mas345).
sr: [00:00.109454] hwdriver: Scan found 0 devices (mastech-ms2115b).
sr: [00:00.109796] hwdriver: Scan found 0 devices (mastech-ms5308).
sr: [00:00.110265] hwdriver: Scan found 0 devices (mastech-ms6514).
sr: [00:00.110579] hwdriver: Scan found 0 devices (mastech-ms8250b).
sr: [00:00.110895] hwdriver: Scan found 0 devices (mastech-ms8250d).
sr: [00:00.111191] hwdriver: Scan found 0 devices (maynuo-m97).
sr: [00:00.111533] hwdriver: Scan found 0 devices (meterman-38xr).
sr: [00:00.111902] hwdriver: Scan found 0 devices (metex-m3640d).
sr: [00:00.112221] hwdriver: Scan found 0 devices (metex-m3860m).
sr: [00:00.112573] hwdriver: Scan found 0 devices (metex-m4650cr).
sr: [00:00.112884] hwdriver: Scan found 0 devices (metex-me21).
sr: [00:00.113230] hwdriver: Scan found 0 devices (metex-me31).
sr: [00:00.113527] hwdriver: Scan found 0 devices (metrix-mx56c).
sr: [00:00.113896] hwdriver: Scan found 0 devices (mic-98581).
sr: [00:00.114324] hwdriver: Scan found 0 devices (mic-98583).
sr: [00:00.114829] serial: No serial device specified.
[... 194 lines cut ...]
Init analog time signal  "Ah1 [Ah]" , signal_start_timestamp_ =  "2021.09.10 17:02:40.388"
sr: [00:00.548791] session: Creating our own main context.
sr: [00:00.548824] session: Starting.
sr: [00:00.548831] hwdriver: itech-it8500: Starting acquisition.
sr: [00:00.548848] session: bus: Received SR_DF_HEADER packet.
Start aquisition for  "ITECH 8511A" ,  aquisition_start_timestamp_ =  "2021.09.10 17:02:40.388"
BaseDevice::close(): Trying to close device  "ITECH 8511A 1.52 4757510274 (/dev/ttyUSB0)"
sr: [00:00.608892] session: Stopping.
sr: [00:00.608909] hwdriver: itech-it8500: Stopping acquisition.
sr: [00:00.608915] session: bus: Received SR_DF_END packet.
sr: [00:00.608932] session: fd_source_finalize: key 0x55a7b5ec6d50
sr: [00:00.608948] session: Stopped.
sr: [00:00.608999] device: itech-it8500: Closing device instance.
sr: [00:00.609010] itech-it8500: itech_it8500_send_cmd: Sending command: 20
sr: [00:00.609188] serial: Wrote 26/26 bytes.
[... cut ...]
sr: [00:00.664820] itech-it8500: itech_it8500_send_cmd: Response packet received: cmd=12
sr: [00:00.664835] serial: Closing serial port /dev/ttyUSB0.
BaseDevice::close(): Device closed  "ITECH 8511A 1.52 4757510274 (/dev/ttyUSB0)"
BaseDevice::~BaseDevice():  "Demo device"
BaseDevice::close(): Trying to close device  "Demo device"
main() failed:  invalid argument
```

I've built the 871df34 commit (which is the current head of the `itech` branch): https://copr.fedorainfracloud.org/coprs/spike/sigrok/builds/