knarfS/smuview #36: OWON P4305 support

ric96, 2021-08-28:
Does smuview support OWON P4305 yet? I only get the following error when trying to select `scpi-pps` device.

```
sr: serial-libsp: Error opening port (25): Inappropriate ioctl for device.
```

knarfS, 2021-08-29:
Yes, the Owon P4305 and P4603 are supported by libsigrok and thus by SmuView.

Please provide more information (as stated in the issue template) to get help, like SmuView version, **full** log output (start SmuView with `-l 5`), operating system, ...

My best guess is, that there is something wrong with an udev rule you have created for your `ttyUSBx`.

ric96, 2021-09-01:
I fixed my udev. So it finds the device but doesn't recognize it.

```
sr: [01:39.553194] scpi_usbtmc: Found USBTMC device (VID:PID = 5345:1235, bus.address = 1.61).
sr: [01:39.553284] scpi: Opening USBTMC device usbtmc/1.61.
sr: [01:39.553339] usb: Found USB device (VID:PID = 5345:1235, bus.address = 1.61).
sr: [01:39.553350] usb: Trying to open USB device 1.61.
sr: [01:39.553406] usb: Opened USB device (VID:PID = 5345:1235, bus.address = 1.61).
sr: [01:39.553415] scpi_usbtmc: Interface 0 configuration 1.
sr: [01:39.553420] scpi_usbtmc: Bulk IN EP 1
sr: [01:39.553428] scpi_usbtmc: Bulk OUT EP 2
sr: [01:39.555323] scpi_usbtmc: Device capabilities: L4, T6, SR0, RL0, DT0
sr: [01:39.555420] scpi_usbtmc: Successfully sent SCPI command: '*IDN?'.
sr: [01:39.555572] scpi: Got response: 'OWON,P4305,2001593,FV:V1.5.0', length 28.
sr: [01:39.558391] usb: Closed USB device 1.61.
sr: [01:39.566579] hwdriver: Scan found 0 devices (scpi-pps).
```
Full Log:

```
[sahaj@athena ~]$ SmuView-0.0.5-x86_64.AppImage -D -d scpi-pps -l 5
sr: [00:00.000000] log: libsigrok loglevel set to 5.
Settings: "/home/sahaj/.config/sigrok/SmuView.conf" format QSettings::NativeFormat
sr: [00:00.001670] backend: libsigrok 0.6.0-git-ba1fb5e/4:0:0.
sr: [00:00.002099] backend: Libs: glib 2.48.2 (rt: 2.68.3/6803:3), libzip 1.0.1, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.24.11584 API 0x01000104, hidapi 0.8.0-rc1, bluez 5.37, libftdi 1.2.
sr: [00:00.002108] backend: Host: x86_64-pc-linux-gnu, little-endian.
sr: [00:00.002116] backend: SCPI backends: TCP, RPC, serial, USBTMC.
sr: [00:00.002120] backend: Firmware search paths:
sr: [00:00.002137] backend:  - /home/sahaj/.local/share/sigrok-firmware
sr: [00:00.002141] backend:  - /home/frank/smuview/build/deps/share/sigrok-firmware
sr: [00:00.002144] backend:  - /home/sahaj/.local/share/flatpak/exports/share/sigrok-firmware
sr: [00:00.002148] backend:  - /var/lib/flatpak/exports/share/sigrok-firmware
sr: [00:00.002152] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.002158] backend:  - /usr/share/sigrok-firmware
sr: [00:00.002170] backend: Sanity-checking all drivers.
sr: [00:00.002178] backend: Sanity-checking all input modules.
sr: [00:00.002183] backend: Sanity-checking all output modules.
sr: [00:00.002187] backend: Sanity-checking all transform modules.
sr: [00:00.074875] scpi_usbtmc: Found USBTMC device (VID:PID = 5345:1235, bus.address = 1.64).
sr: [00:00.074913] scpi: Opening USBTMC device usbtmc/1.64.
sr: [00:00.075023] usb: Found USB device (VID:PID = 5345:1235, bus.address = 1.64).
sr: [00:00.075039] usb: Trying to open USB device 1.64.
sr: [00:00.075093] usb: Opened USB device (VID:PID = 5345:1235, bus.address = 1.64).
sr: [00:00.075104] scpi_usbtmc: Interface 0 configuration 1.
sr: [00:00.075116] scpi_usbtmc: Bulk IN EP 1
sr: [00:00.075123] scpi_usbtmc: Bulk OUT EP 2
sr: [00:00.076756] scpi_usbtmc: Device capabilities: L4, T6, SR0, RL0, DT0
sr: [00:00.076839] scpi_usbtmc: Successfully sent SCPI command: '*IDN?'.
sr: [00:00.076993] scpi: Got response: 'OWON,P4305,2001593,FV:V1.5.0', length 28.
sr: [00:00.080054] usb: Closed USB device 1.64.
sr: [00:00.097719] hwdriver: Scan found 0 devices (scpi-pps).
```

ric96, 2021-09-01:
aha! libsirok included owon support much recently than i thought. It is working now :D

closed 2021-09-01 by ric96

knarfS, 2021-09-02:
Basic support for the Owon supplies was added in March with Commit https://github.com/sigrokproject/libsigrok/commit/5433907ed99c40e948b8350f55845dd88a5cbbe6

SmuView 0.0.5 doesn't include the driver for these supplies, but the continuous build of SmuView does.