knarfS/smuview #39: smuview "crashes" the UT-D04 (USB/HID cable with WCH CH9325 chip, USB VID/PID 1a86:e008)

fhvyhjriur, 2021-11-26:
This is somewhat interesting issue i think.
I know about issue #31 . 

This is working perfectly fine: 
1. connect the usb cable to computer
2. run: sigrok-cli -d voltcraft-vc840:conn=1a86.e008 --continuous

smuview does not work at all with my hardware. I am running SmuView-0.0.6-git-7050317-x86_64.AppImage .
Its crashing the WCH chipset when its been started.
This is the result:
./SmuView-0.0.6-git-7050317-x86_64.AppImage
sr: testo: Failed to get manufacturer string descriptor: LIBUSB_ERROR_IO.
sr: testo: Failed to get product string descriptor: LIBUSB_ERROR_IO.

Then i close smuview and run:
sigrok-cli -d voltcraft-vc840:conn=1a86.e008 --continuous
sr: uni-t-dmm: HID feature report error: LIBUSB_ERROR_IO.
sr: uni-t-dmm: HID chip init failed: -1.

I have to disconnect the usb cable and connect it again to get a working sigrok-cli again.
Running 
./SmuView-0.0.6-git-7050317-x86_64.AppImage -d voltcraft-vc840:conn=1a86.e008 
does not change anything. I still get the same Error in the command line and no data from the DMM.

My conclusion without reading the smuview code is that smuview does something at the start that crashes the WCH chip. Then smuview and sigrok-cli are not able to read any data from it until reconnect.

knarfS, 2021-12-17:
Could you please start SmuView (and sigrok-cli) with the option `-l 5` and provide the output. Maybe this will give a hint what the problem is.

fhvyhjriur, 2021-12-20:
sigrok-cli -d voltcraft-vc840:conn=1a86.e008 -l 5 --continuous
sr: [00:00.000002] log: libsigrok loglevel set to 5.
sr: [00:00.000063] backend: libsigrok 0.5.2/5:1:1.
sr: [00:00.000287] backend: Libs: glib 2.67.5 (rt: 2.68.1/6801:1), libzip 1.7.3, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.24.11584 API 0x01000108, hidapi 0.10.1, bluez 5.56, libftdi 1.5.
sr: [00:00.000311] backend: Host: x86_64-pc-linux-gnu, little-endian.
sr: [00:00.000440] backend: SCPI backends: TCP, serial, USBTMC.
sr: [00:00.000453] backend: Firmware search paths:
sr: [00:00.000648] backend:  - /home/user/.local/share/sigrok-firmware
sr: [00:00.000665] backend:  - /usr/share/sigrok-firmware
sr: [00:00.000677] backend:  - /usr/share/plasma/sigrok-firmware
sr: [00:00.000779] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.000790] backend:  - /usr/share/sigrok-firmware
sr: [00:00.000891] backend:  - /var/lib/snapd/desktop/sigrok-firmware
sr: [00:00.001011] backend: Sanity-checking all drivers.
sr: [00:00.001034] backend: Sanity-checking all input modules.
sr: [00:00.001128] backend: Sanity-checking all output modules.
sr: [00:00.001280] backend: Sanity-checking all transform modules.
srd: libsigrokdecode loglevel set to 5.
sr: [00:00.011279] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi (nil) cg NULL -> [uint32 20000]
sr: [00:00.011487] usb: Found USB device (VID:PID = 1a86:e008, bus.address = 3.3).
sr: [00:00.011518] hwdriver: Scan found 1 devices (voltcraft-vc840).
sr: [00:00.011694] device: voltcraft-vc840: Opening device instance.
sr: [00:00.011713] usb: Trying to open USB device 3.3.
sr: [00:00.011850] usb: Opened USB device (VID:PID = 1a86:e008, bus.address = 3.3).
sr: [00:00.011971] session: Using thread-default main context.
sr: [00:00.011990] session: Starting.
sr: [00:00.012002] hwdriver: voltcraft-vc840: Starting acquisition.
sr: [00:00.012102] session: bus: Received SR_DF_HEADER packet.
cli: Received SR_DF_HEADER.
Press any key to stop acquisition.
sr: [00:00.022966] uni-t-dmm: Sending initial HID feature report.
sr: [00:00.023027] uni-t-dmm: HID init = 0x60 0x09 0x00 0x00 0x03 (2400 baud)
sr: [00:00.028909] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.040785] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.052799] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.064833] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.079229] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.088814] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.100819] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.112806] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.124824] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.136814] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.148815] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.160804] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.172802] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.184806] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.196806] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.208801] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.220801] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.232789] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.244807] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.256787] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.268800] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.280805] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.292807] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.304805] uni-t-dmm: 8-byte chunk: f0 00 00 00 00 00 00 00 (0 data bytes)
sr: [00:00.316803] uni-t-dmm: 8-byte chunk: f1 13 00 00 00 00 00 00 (1 data bytes)
sr: [00:00.320824] uni-t-dmm: 8-byte chunk: f1 20 00 00 00 00 00 00 (1 data bytes)
sr: [00:00.328790] uni-t-dmm: 8-byte chunk: f1 35 00 00 00 00 00 00 (1 data bytes)
sr: [00:00.340824] uni-t-dmm: 8-byte chunk: f1 47 00 00 00 00 00 00 (1 data bytes)
sr: [00:00.352802] uni-t-dmm: 8-byte chunk: f3 5d 68 75 00 00 00 00 (3 data bytes)
sr: [00:00.364804] uni-t-dmm: 8-byte chunk: f3 81 95 a0 00 00 00 00 (3 data bytes)
sr: [00:00.372796] uni-t-dmm: 8-byte chunk: f2 b2 c4 00 00 00 00 00 (2 data bytes)
sr: [00:00.384794] uni-t-dmm: 8-byte chunk: f2 d0 e8 00 00 00 00 00 (2 data bytes)
sr: [00:00.384855] uni-t-dmm: DMM packet:   13 20 35 47 5d 68 75 81 95 a0 b2 c4 d0 e8
sr: [00:00.384898] fs9721: Digits: 05 7d 05 15 (1017).
sr: [00:00.384941] fs9721: Decimal point after second digit.
sr: [00:00.384980] fs9721: The display value is 1017.000000.
sr: [00:00.385061] fs9721: RS232 enabled.
sr: [00:00.385102] fs9721: User-defined LCD symbol 3 is active.
[... 175 lines cut ...]
sr: [00:00.781771] hwdriver: Scan found 0 devices (voltcraft-vc940).
sr: [00:00.782848] hwdriver: Scan found 0 devices (voltcraft-vc940-ser).
sr: [00:00.783915] hwdriver: Scan found 0 devices (voltcraft-vc96).
sr: [00:00.784936] hwdriver: Scan found 0 devices (voltcraft-vc960).
[... cut ...]
sr: [00:00.797244] session: bus: Received SR_DF_HEADER packet.
Start aquisition for  "Voltcraft VC-840" ,  aquisition_start_timestamp_ =  "2021.12.20 19:38:34.886"
sr: [00:00.807562] uni-t-dmm: Sending initial HID feature report.
sr: [00:00.807616] uni-t-dmm: HID init = 0x60 0x09 0x00 0x00 0x03 (2400 baud)
sr: [00:00.809244] uni-t-dmm: HID feature report error: LIBUSB_ERROR_IO.
sr: [00:00.809288] uni-t-dmm: HID chip init failed: -1.
sr: [00:00.809329] session: fd_source_finalize: key 0xffffffffffffffff
sr: [00:00.809388] session: Stopped.

fhvyhjriur, 2022-06-07:
I like to mention, that i still like to test possible fixes for this issue or something else that could help me to get able to use smuview.

fhvyhjriur, 2022-11-01:
After about a year after the bugreport and the issue still existing on even different Linux-Distro, i gave up and got a https://sigrok.org/wiki/Device_cables#UNI-T_UT-D02 cable and use it with a usb-rs232 adapter. I use now 
sigrok-cli -d voltcraft-vc840-ser:conn=/dev/ttyUSB0 --continuous . sigrok-cli -d voltcraft-vc840-ser:conn=/dev/ttyUSB0 --scan also detect the DMM without issues.
I can also use SmuView-0.0.6-git-7050317-x86_64.AppImage . It looks like its detecting the DMM the same way sigrok-cli -d voltcraft-vc840-ser:conn=/dev/ttyUSB0 --scan does. Its important to leave the baudrate in the SmuView gui free and not choose any of the listed possible baudrate in the dropdown. The reason is, because SmuView does not list 2400 as a possible baudrate in the dropdown list. From the detailed smuview log i can see: serial: Got params: rate 2400, frame 8/0/1, flow 0, rts 0, dtr 1.
The most simple is to start it directly with 
./SmuView-0.0.6-git-7050317-x86_64.AppImage -d voltcraft-vc840-ser:conn=/dev/ttyUSB0

Then it detects it and starts working without the need of any configuration in the gui.

netweaver1970, 2023-08-20:
> After about a year after the bugreport and the issue still existing on even different Linux-Distro, i gave up and got a https://sigrok.org/wiki/Device_cables#UNI-T_UT-D02 cable and use it with a [...]

thanks for this post, it helped me indeed getting my UT61E working on MacOS Ventura with SMUVIEW. I had already a serial cable+serial2USB convertor (and had it working with the CLI) but I have an USB cable version on the way, thinking USB might be easier. I guess I'll have to drop the USB cable, when it finally arrives. 
Btw, for reference, the serial port created this way in MacOS is /dev/tty.usbserial-120