knarfS/smuview #40: continuous build won't run on debian sid

folkertvanheusden, 2021-11-28:
Running it gives:

 /tmp/.mount_SmuVienaCk2D/usr/bin/smuview: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0: undefined symbol: g_module_open_full

That's for: SmuView-0.0.5-x86_64.AppImage and SmuView-0.0.6-git-7050317-x86_64.AppImage‎

knarfS, 2021-12-17:
That's because of the new glib 2.70 :(
Here you can find a very good analysis of the problem: https://github.com/project-slippi/Ishiiruka/issues/323

As workaround for now, you can start SmuView like this:
```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgmodule-2.0.so ./SmuView-0.0.6-git-7050317-x86_64.AppImage
```

I'm working on a PR that will fix this problem in the AppImage tools itself.

knarfS, 2022-11-29:
Fixed in the current continuous build: https://github.com/knarfS/smuview/releases/tag/continuous

closed 2022-11-29 by knarfS