knarfS/smuview #48: Nightly smuview AppImage refuses to run on Ubuntu 22.04

jarricd, 2022-10-20:
**Describe the bug**
Nightly smuview AppImage refuses to run on Ubuntu 22.04 due to this error:
`(base) damian@damian-GT72VR-7RD:~/smuview$ ./SmuView-0.0.6-git-7050317-x86_64.AppImage 
/tmp/.mount_SmuViexDp88T/usr/bin/smuview: symbol lookup error: /lib/x86_64-linux-gnu/libgio-2.0.so.0: undefined symbol: g_module_open_full
(base) damian@damian-GT72VR-7RD:~/smuview$ 
`
I can't quite make out what I need to install on my system to run the AppImage.

**To Reproduce**
Steps to reproduce the behavior:
1. Download smuview specified in the previous step.
2. chmod +x <filename>
3. Try to run smuview by running ./<filename>
4. Observe the error.

**Expected behavior**
smuview should run.

**Enviroment (please complete the following information):**
 -  SmuView-0.0.6-git-7050317-x86_64
 - Appimage
 - Ubuntu 22.04 running on Wayland

**Additional context**
More OS info:
`(base) damian@damian-GT72VR-7RD:~/smuview$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy
`
`(base) damian@damian-GT72VR-7RD:~/smuview$ uname -a
Linux damian-GT72VR-7RD 5.15.0-50-generic #56-Ubuntu SMP Tue Sep 20 13:23:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
Anaconda base environment is always activated due to the type of work I do on this host.

mrv96, 2022-11-28:
This workaround should work:

```bash
./my_file.AppImage --appimage-extract
rm -rf squashfs-root/usr/lib/libgmodule*
wget "https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage"
chmod a+x appimagetool-x86_64.AppImage
./appimagetool-x86_64.AppImage -v squashfs-root # -n option may be required
rm -rf squashfs-root/
```
Taken from: https://github.com/project-slippi/Ishiiruka/issues/323#issuecomment-977415376

knarfS, 2022-11-28:
This is already fixed in Upstream with https://github.com/AppImageCommunity/pkg2appimage/commit/1e3ecde6b92ef198405463415b8090a4435d0cb0 (also see https://github.com/AppImageCommunity/pkg2appimage/pull/500), but the SmuView build pipeline is failing at the moment, because of the stupid macOS build :)

I'm working on fixing the build process right now. But maybe i have to disable the macOS build to have downloads for Linux and Windows...

knarfS, 2022-11-29:
Fixed in the current continuous build: https://github.com/knarfS/smuview/releases/tag/continuous