knarfS/smuview #51: can't connect to devices

kjuh78, 2022-12-03:
**Describe the bug**
SmuView can't connect to either my DMM (Brymen BM869s, connected via BM-86X), nor my PSU (RND 320-KA3005P, connected via USB). Both are detected by sigrok-cli (0.8.0-git-525f481).

**To Reproduce**
Steps to reproduce the behavior:
1. connect device (DMM: Brymen BM869s via BM-86X USB interface cable; PSU: RND 320-K3005P via USB cable)
2. add device in smuview 
3. specify connection details
DMM: 
driver: brymen-bm86x, interface: USB
PSU:
driver: korad-kaxxxxp; interface: serial; port: /dev/cu.usbmodem0029128002411 (USB Virtual COM)
4. click "scan for devices using driver above"

**Expected behavior**
At least the DMM should be detected and listed, the PSU might not be recognized as compatible by the driver.

**Screenshots or log**

**Enviroment (please complete the following information):**
 - SmuView 0.0.5-git-3193851
 - nightly DMG
 - OS: MacOS 12.6.1 on MacBook Pro M1 2021

**Additional context**
% /Applications/sigrok-cli.app/Contents/MacOS/sigrok-cli --scan                                                           
The following devices were found:
demo - Demo device with 13 channels: D0 D1 D2 D3 D4 D5 D6 D7 A0 A1 A2 A3 A4
brymen-bm86x - Brymen BM86x with 2 channels: P1 P2

% /Applications/sigrok-cli.app/Contents/MacOS/sigrok-cli --driver=korad-kaxxxxp:conn=/dev/tty.usbmodem0029128002411 --scan
sr: korad-kaxxxxp: Unknown model ID 'RND 320-KA3005P V4.1' detected, aborting.

% /Applications/sigrok-cli.app/Contents/MacOS/sigrok-cli --version 
sigrok-cli 0.8.0-git-525f481

Libraries and features:
- libsigrok 0.6.0-git-6dc55e4/4:0:0 (rt: 0.6.0-git-6dc55e4/4:0:0).
 - Libs:
  - glib 2.64.2 (rt: 2.64.2/6402:2)
  - zlib 1.2.5
  - libzip 1.6.1
  - minilzo 2.10
  - libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1)
  - libusb-1.0 1.0.23.11397 API 0x01000107
  - hidapi 0.9.0
  - libftdi 1.4
  - Host: x86_64-apple-darwin15.6.0, little-endian.
  - SCPI backends: TCP, RPC, serial, USBTMC.
- libsigrokdecode 0.6.0-git-24ba9e1/4:0:0 (rt: 0.6.0-git-24ba9e1/4:0:0).
 - Libs:
  - glib 2.64.2 (rt: 2.64.2/6402:2)
  - Python 3.7.7 / 0x30707f0 (API 1013, ABI 3)
  - Host: x86_64-apple-darwin15.6.0, little-endian.

knarfS, 2022-12-03:
SmuView 0.0.5-git-3193851 is very old, please try the current [continuous build](https://github.com/knarfS/smuview/releases/tag/continuous), which also has a much newer libsigrok (0.6.0-git-8c08409).

You can also use the same sigrok-cli argument for connecting devices:
```
./smuview --driver=korad-kaxxxxp:conn=/dev/tty.usbmodem0029128002411
``

knarfS, 2022-12-03:
It would be also helpful to start SmuView with the `-l 5` argument and append the log output.

kjuh78, 2022-12-03:
Sorry. Didn't realize I'm using an old version, it's still listed as current stable on https://sigrok.org/wiki/SmuView#macOS .
The current nightly won't start with python 3.11 installed (INSTALL states >=3 as requirement), because it can't find python 3.10:
```
% /Applications/SmuView.app/Contents/MacOS/smuview -l 5
dyld[8666]: Library not loaded: '/usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/Python'
  Referenced from: '/Applications/SmuView.app/Contents/MacOS/smuview.real'
  Reason: tried: '/usr/local/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/Python' (no such file), '/Library/Frameworks/Python.framework/Versions/3.10/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.10/Python' (no such file)
zsh: abort      /Applications/SmuView.app/Contents/MacOS/smuview -l 5
```
Creating a symlink from /Library/Frameworks/Python.framework/Versions/3.10 to /Library/Frameworks/Python.framework/Versions/3.11 seems to help for now.
Is this a typo in INSTALL or does the App request the wrong version?
Should I install python 3.10 or is INSTALL right?

Here's the requested log (created by SmuView 0.0.6-git-54cf16b):
```
% /Applications/SmuView.app/Contents/MacOS/smuview -l 5
Settings: "/Users/stefan/Library/Preferences/org.sigrok.SmuView.plist" format QSettings::NativeFormat
sr: [00:00.000001] log: libsigrok loglevel set to 5.
sr: [00:00.031944] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.053453] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.198395] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.202522] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.204167] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.205281] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.206753] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.209239] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:00.209255] hwdriver: Scan found 0 devices (brymen-bm52x).
sr: [00:00.210274] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.211727] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:00.211733] hwdriver: Scan found 0 devices (brymen-bm82x).
sr: [00:00.212458] hwdriver: Scan found 0 devices (brymen-bm85x).
sr: [00:00.213256] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.214584] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:00.214590] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:00.215618] hwdriver: Scan found 0 devices (cem-dt-885x).
sr: [00:00.216477] hwdriver: Scan found 0 devices (center-309).
sr: [00:00.217274] hwdriver: Scan found 0 devices (colead-slm).
sr: [00:00.217992] hwdriver: Scan found 0 devices (conrad-digi-35-cpu).
sr: [00:00.218746] serial: No serial device specified.
sr: [00:00.220016] hwdriver: Scan found 0 devices (dcttech-usbrelay).
sr: [00:00.220727] demo: Generating square pattern.
sr: [00:00.220736] demo: Generating sine pattern.
sr: [00:00.220753] demo: Generating triangle pattern.
sr: [00:00.220792] demo: Generating sawtooth pattern.
sr: [00:00.220805] hwdriver: Scan found 1 devices (demo).
sr: [00:00.221868] hwdriver: Scan found 0 devices (deree-de5000).
sr: [00:00.222788] hwdriver: Scan found 0 devices (digitek-dt4000zc).
sr: [00:00.224230] hwdriver: Scan found 0 devices (eevblog-121gw).
sr: [00:00.225165] serial: No serial device specified.
sr: [00:00.236123] hwdriver: Scan found 0 devices (fluke-45).
sr: [00:00.237168] hwdriver: Scan found 0 devices (fluke-dmm).
sr: [00:00.237973] hwdriver: Scan found 0 devices (gmc-mh-1x-2x-rs232).
sr: [00:00.238808] hwdriver: Scan found 0 devices (gmc-mh-2x-bd232).
sr: [00:00.239833] hwdriver: Scan found 0 devices (gwinstek-gdm-397).
sr: [00:00.240986] serial: No serial device specified.
sr: [00:00.251933] hwdriver: Scan found 0 devices (gwinstek-gds-800).
sr: [00:00.252859] hwdriver: Scan found 0 devices (gwinstek-gpd).
sr: [00:00.253700] serial: No serial device specified.
sr: [00:00.264494] hwdriver: Scan found 0 devices (hameg-hmo).
sr: [00:00.266294] hwdriver: Scan found 0 devices (hantek-6xxx).
sr: [00:00.267472] hwdriver: Scan found 0 devices (hantek-dso).
sr: [00:00.268351] serial: No serial device specified.
sr: [00:00.268359] hwdriver: Scan found 0 devices (hp-3457a).
sr: [00:00.269087] serial: No serial device specified.
sr: [00:00.269091] hwdriver: Scan found 0 devices (hp-59306a).
sr: [00:00.270039] serial: No serial device specified.
sr: [00:00.270044] hwdriver: Scan found 0 devices (hpib-pps).
sr: [00:00.270739] hwdriver: Scan found 0 devices (iso-tech-idm103n).
sr: [00:00.271434] hwdriver: Scan found 0 devices (itech-it8500).
sr: [00:00.272744] hwdriver: Scan found 0 devices (kecheng-kc-330b).
[... 117 lines cut ...]
sr: [00:00.464769] hwdriver: Scan found 0 devices (voltcraft-vc830-ser).
sr: [00:00.465479] hwdriver: Scan found 0 devices (voltcraft-vc840).
sr: [00:00.466146] hwdriver: Scan found 0 devices (voltcraft-vc840-ser).
sr: [00:00.466774] hwdriver: Scan found 0 devices (voltcraft-vc870).
sr: [00:00.467389] hwdriver: Scan found 0 devices (voltcraft-vc870-ser).
sr: [00:00.468001] hwdriver: Scan found 0 devices (voltcraft-vc920).
sr: [00:00.468617] hwdriver: Scan found 0 devices (voltcraft-vc920-ser).
sr: [00:00.469233] hwdriver: Scan found 0 devices (voltcraft-vc940).
sr: [00:00.469837] hwdriver: Scan found 0 devices (voltcraft-vc940-ser).
[... cut ...]
sr: [00:10.860065] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:18.746261] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:18.749255] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:18.749270] hwdriver: Scan found 0 devices (brymen-bm86x).
sr: [00:23.963010] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:23.965265] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
sr: [00:23.965282] hwdriver: Scan found 0 devices (brymen-bm86x).

```

kjuh78, 2022-12-03:
BTW: connecting to my 2nd PSU of the same type (RND 320-KA3005P) works fine. Obviously this one has an older FW version and is thus correctly identified. I'll report the identification string to libsigrok.

knarfS, 2022-12-11:
Your `sigrok-cli` version is also pretty old (from mid of March 2022). But unfortunately it looks like the nightly build for macOS isn't running at the moment.
A few month ago, there was a change in libsigrok and now more of the korad-kaxxxxp power supplies are recognized out of the box, without the need to specify every version for a new device in the source code.

So both of your korad-kaxxxxp supplies should be found by the continuous build of SmuView and should work out of the box:
```
./smuview -l 5 -d korad-kaxxxxp:conn=/dev/some.serial.connection.string
```
Of course you have to use your specific `/dev/...` string, the `-l 5` option will give you (hopefully) a detailed log output when something goes wrong.

If you want to test it with a current `sigrok-cli` you can use homebrew and install the HEAD-version of `sigrok-cli`:
```
brew install --HEAD sigrok-cli
```
or
```
brew reinstall --HEAD sigrok-cli
```
Full disclosure: I don't have any clue about homebrew ;)

If your RND models aren't recognized for some reason, the `korad-kaxxxxp` driver has a special option, where you can force a specific model to be used. For example using the Korad KA3005P model:
```
./smuview -l 5 -d korad-kaxxxxp:conn=/dev/some.serial.connection.string:force_detect=KORADKA3005PV2.0
```

Regarding you Brymen BM86x cable, it is using HID to connect (if I'm not mistaken). You can see some error when trying to open the HID-device:
```
sr: [00:00.206753] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.209239] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4294994192
```
Not sure what that means, but basically you should be able to connect exactly the same way like when using `sigrok-cli`. When this is working:
```
./sigrok-cli -l 5 -d brymen-bm86x --samples 2
```
then this should also work:
```
./smuview -l 5 -d brymen-bm86x
```
Maybe you can try adding the `vendor.product` to the connect string. Example, not sure if the vendor/product ids are right (`lsusb -v` to check):
```
./smuview -l 5 -d brymen-bm86x:conn=0820.0001
```

Hope that helps, if you still ran into errors, just post the log output with with some explanations what you've done.

kjuh78, 2022-12-12:
The PSUs work fine with the latest build.
Still no luck with the Brymen:

sigrok-cli  seems to work:
```
% sigrok-cli -l 5 -d brymen-bm86x --samples 2
sr: [00:00.000002] log: libsigrok loglevel set to 5.
sr: [00:00.000037] backend: libsigrok 0.6.0-git-6dc55e4/4:0:0.
sr: [00:00.000079] backend: Libs: glib 2.64.2 (rt: 2.64.2/6402:2), zlib 1.2.5, libzip 1.6.1, minilzo 2.10, libserialport 0.1.1/1:0:1 (rt: 0.1.1/1:0:1), libusb-1.0 1.0.23.11397 API 0x01000107, hidapi 0.9.0, libftdi 1.4.
sr: [00:00.000085] backend: Host: x86_64-apple-darwin15.6.0, little-endian.
sr: [00:00.000090] backend: SCPI backends: TCP, RPC, serial, USBTMC.
sr: [00:00.000093] backend: Firmware search paths:
sr: [00:00.000175] backend:  - ../share/sigrok-firmware
sr: [00:00.000201] backend:  - /Users/stefan/.local/share/sigrok-firmware
sr: [00:00.000205] backend:  - /Users/jenkins_slave/sr_macosx/share/sigrok-firmware
sr: [00:00.000211] backend:  - /usr/local/share/sigrok-firmware
sr: [00:00.000214] backend:  - /usr/share/sigrok-firmware
sr: [00:00.000259] backend: Sanity-checking all drivers.
sr: [00:00.000270] backend: Sanity-checking all input modules.
sr: [00:00.000277] backend: Sanity-checking all output modules.
sr: [00:00.000282] backend: Sanity-checking all transform modules.
srd: libsigrokdecode loglevel set to 5.
sr: [00:00.010455] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.015977] serial: Flushing serial port hid/bu86x.
sr: [00:00.015989] serial-dmm: Probing serial port hid/bu86x.
sr: [00:00.017648] serial: Wrote 4/4 bytes.
sr: [00:00.017666] serial: Detecting packets on hid/bu86x (timeout = 3000ms).
sr: [00:01.034929] serial: Read 1/1 bytes.
...
sr: [00:01.036277] serial: Read 1/1 bytes.
sr: [00:01.036325] serial: Trying packet: len 24, bytes 00 00 02 be be a0 fe db 1e 40 40 40 40 40 00 00 86 86 86 86 00 00 00 00
sr: [00:01.036356] serial: Valid packet after 1018ms.
sr: [00:01.036374] serial: RX count 24, packet len 24.
sr: [00:01.036384] serial-dmm: Found device on port hid/bu86x.
sr: [00:01.036879] serial: Closing serial port hid/bu86x.
sr: [00:01.038758] hwdriver: Scan found 1 devices (brymen-bm86x).
sr: [00:01.039487] device: brymen-bm86x: Opening device instance.
sr: [00:01.039522] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:01.045857] serial: Flushing serial port hid/bu86x.
sr: [00:01.046279] hwdriver: sr_config_set(): key 50001 (limit_samples) sdi 0x600001b48000 cg NULL -> uint64 2
sr: [00:01.046372] session: Using thread-default main context.
sr: [00:01.046380] session: Starting.
sr: [00:01.046386] hwdriver: brymen-bm86x: Starting acquisition.
sr: [00:01.046439] session: bus: Received SR_DF_HEADER packet.
cli: Received SR_DF_HEADER.
sr: [00:01.057946] serial-dmm: Requesting next packet.
sr: [00:01.059677] serial: Wrote 4/4 bytes.
sr: [00:01.068929] serial-dmm: Not re-requesting yet, 489ms left.
...
sr: [00:01.448925] serial-dmm: Not re-requesting yet, 109ms left.
sr: [00:01.482413] serial: Read 24/256 bytes.
sr: [00:01.482548] serial-dmm: Checking: pos 0, len 24.
sr: [00:01.482563] serial-dmm: Valid packet, size 24, processing
sr: [00:01.482606] serial-dmm: DMM packet: 00 00 02 be be a0 fe db 1e 40 40 40 40 40 00 00 86 86 86 86 00 00 00 00
sr: [00:01.484063] session: bus: Received SR_DF_ANALOG packet (1 samples).
cli: Received SR_DF_ANALOG (1 samples).
P1: 18.2 °C
sr: [00:01.485209] brymen-bm86x: invalid float string: '----'
sr: [00:01.485228] session: bus: Received SR_DF_ANALOG packet (1 samples).
cli: Received SR_DF_ANALOG (1 samples).
P2: 18 °C
sr: [00:01.485387] serial-dmm: Not re-requesting yet, 100ms left.
...
sr: [00:01.584829] serial-dmm: Not re-requesting yet, 0ms left.
sr: [00:01.595950] serial-dmm: Requesting next packet.
sr: [00:01.597850] serial: Wrote 4/4 bytes.
sr: [00:01.606695] serial-dmm: Not re-requesting yet, 489ms left.
...
sr: [00:02.091037] serial-dmm: Not re-requesting yet, 4ms left.
sr: [00:02.104414] serial-dmm: Requesting next packet.
[... 50 lines cut ...]
specifying VID/PID (I checked, they are correct), doesn't make any difference:
```
% smuview -l 5 -d brymen-bm86x:conn=0820.0001
Settings: "/Users/stefan/Library/Preferences/org.sigrok.SmuView.plist" format QSettings::NativeFormat
sr: [00:00.000001] log: libsigrok loglevel set to 5.
sr: [00:00.029686] hwdriver: Scan found 0 devices (agilent-dmm).
sr: [00:00.048736] hwdriver: Scan found 0 devices (appa-55ii).
sr: [00:00.176694] hwdriver: Scan found 0 devices (arachnid-labs-re-load-pro).
sr: [00:00.180668] hwdriver: Scan found 0 devices (atten-pps3203).
sr: [00:00.181919] hwdriver: Scan found 0 devices (bbcgm-2010).
sr: [00:00.182881] hwdriver: Scan found 0 devices (brymen-bm25x).
sr: [00:00.183617] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.186014] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4295153566
sr: [00:00.186022] hwdriver: Scan found 0 devices (brymen-bm52x).
sr: [00:00.189546] serial: Opening serial port 'hid/bu86x' (flags 1).
sr: [00:00.191044] serial-hid: Unsupported HIDAPI path format: DevSrvsID:4295153566
sr: [00:00.191050] hwdriver: Scan found 0 devices (brymen-bm82x).
sr: [00:00.191811] hwdriver: Scan found 0 devices (brymen-bm85x).
...
sr: [00:00.470103] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi 0x0 cg NULL -> [uint32 20000, 20001]
sr: [00:00.471177] hwdriver: sr_config_list(): key 2147418112 (NULL) sdi 0x0 cg NULL -> [uint32 20000, 20001]
sr: [00:00.471193] serial: Opening serial port '0820.0001' (flags 1).
sr: [00:00.471349] serial-libsp: Error getting port from name 0820.0001: (2) No such file or directory.
sr: [00:00.471353] hwdriver: Scan found 0 devices (brymen-bm86x).
```

knarfS, 2022-12-13:
Ok, here is my last idea:
```
./smuview -l 5 -d brymen-bm86x:conn=hid/bu86x
```

For reference : https://github.com/sigrokproject/libsigrok/blob/master/README.devices, chapter "Specifying serial ports"

Looks like that there is a platform specific problem. Here I found a similar bug related to `hidapi` and macOS: https://sigrok.org/bugzilla/show_bug.cgi?id=1586
Maybe `sigrok-cli` and SmuView are using different library version. Unfortunately SmuView doesn't print the used library versions at startup. I'll fix that, it is also important for the Python problem.