knarfS/smuview #53: Unwanted rounding in multimeter display

p1ngb4ck, 2022-12-12:
I use a UT61C multimeter connected via RS232.
When I test a resistor of 4.7k, I get a display of 5k in the main display, while the graphs and sigrok-cli clearly show the correct values. There is some unwanted rounding active on the display routine ..

knarfS, 2022-12-12:
Some more information would be helpful:
* What SmuView version are you using?
* What was the expected value?
* What was the display exactly showing (screenshot)
* When you start SmuView with the argument `-l 5` does the right value appear in the log? If so, please attach this portion of the log (with maybe 10 lines before and after the interesting part)

knarfS, 2022-12-16:
I think I've found the problem. SmuView is using the `encoding.digits`/`spec.spec_digits` information, provided in each individual analog data payload by the libsigrok drivers. This information is used to get a representation of the measured value as close as possible to the display of the DMM.

It looks like the [fs9922 protocol decoder](https://github.com/sigrokproject/libsigrok/blob/master/src/dmm/fs9922.c) used for the UT61C is setting the exponent rather the number of decimal places. From the docstring in [`libsigrok.h`](https://github.com/sigrokproject/libsigrok/blob/49c45a5c64a031c4371710cad1e7a63c9d67c318/include/libsigrok/libsigrok.h#L531):
> Number of significant digits after the decimal point if positive,
> or number of non-significant digits before the decimal point if
> negative [...]

I've fixed that issue [in my branch `fs9922_digits`](https://github.com/knarfS/libsigrok/tree/fs9922_digits), but of course couldn't test it. Give it a try if you like. Would be great, if you could test also for other decimal places and prefixes (µ, m, k, M)- Let me know, if this works for you, then I'll open a PR in the next days.

p1ngb4ck, 2022-12-17:
This works absolutely perfect so far, thank you soo much. I get exactly whats on the display of my UT-61C now.
I tested any resistors I had handy (k / M works fine), as well as some defined voltages (160mV, -12V, 3.3V, 5V ,12V, 40V)
Signed ( - ) numbers/decimals also work just fine.
Will test some caps over the weekend to check µF & nF.

Thanks sooooo  much for your work, this is just incredible =) It allows me to integrate my multimeter (and hopefully psu too soon) in my videos/streams via OBS in linux with a nice display !! Just great.

p1ngb4ck, 2022-12-23:
I can confirm now, that all modes and decimal places and units work just fine.

knarfS, 2023-02-25:
The wrong value inside the main display is fixed with commit https://github.com/knarfS/smuview/commit/1be2c391ecef3073e8613ecb4cc31a8dfd73941c and the latest [continuous build](https://github.com/knarfS/smuview/releases/tag/continuous).

closed 2023-02-25 by knarfS