knarfS/smuview #54: Displaying error in Panel View

matika77, 2022-12-20:
Hello,

I got an UNI-T UT61E Multimeter with a HID USB cable and want to use SmuView. After some Time having Problems to let SmuView recognize my Multimeter (can not be added by "Add device"), I got it to work with "SmuView-0.0.6-git-54cf16b-x86_64.AppImage --driver=uni-t-ut61e:conn=1a86.e008".

It works, but the Panel View is giving me wrong results when the Ammeter is measuring below 1mA. It just cuts off the first digit.

[image]

The other Views show the Value correctly.

System: X86-64, Kubuntu 20.04, tried SmuView-git and SmuView 0.0.5.

Thank you for the good work :-)

knarfS, 2022-12-21:
Hey there,

> After some Time having Problems to let SmuView recognize my Multimeter (can not be added by "Add device"), I got it to work with "SmuView-0.0.6-git-54cf16b-x86_64.AppImage --driver=uni-t-ut61e:conn=1a86.e008".

libsigrok (and therefor SmuView) can't scan for serial devices, because just trying to talk to them could result in unexpected behavior. When using the "Add device" dialog, SmuView populates the "Serial Port" drop down selector with potential serial ports it has found. But there is a good chance, that some of them are missing.
If so, you have to enter the correct connecting string by yourself into the drop down (e.g. `1a86.e008` in your case) or use the `--driver` / `-d` parameter at startup.
Have a look at libsigroks [README.devices](https://github.com/sigrokproject/libsigrok/blob/master/README.devices) for more info about connecting serial devices.

Regarding the wrong results in the panel view, this is closely related to https://github.com/knarfS/smuview/issues/53. Right now, I'm in the process of fixing that.

> Thank you for the good work :-)

Thanks :)

knarfS, 2023-02-25:
The wrong value inside the Panel View is fixed with commit https://github.com/knarfS/smuview/commit/1be2c391ecef3073e8613ecb4cc31a8dfd73941c and the latest [continuous build](https://github.com/knarfS/smuview/releases/tag/continuous).

closed 2023-02-25 by knarfS