knarfS/smuview #55: Build error: sigrok::Quantity::ENERGY and similar not found

TilCreator, 2022-12-24:
I'm trying to build the project and got pretty far, but the build itself fails with a bunch of errors.
[complete build log and dependencies list](https://gist.github.com/TilCreator/765ac11a5bac164382c9c33b197936cb)

It fails to find the type `sigrok::Quantity::ENERGY`, I also did not find `ENERGY` in the [0.5.2 API docs](https://www.sigrok.org/api/libsigrok/0.5.1/bindings/cxx/a00701.html)

Is a special fork of libsigrok required, I did not find anything in the readme or the INSTALL docs?

knarfS, 2022-12-25:
You are trying to build SmuView 0.0.5 with libsigrok 0.5.2. Both releases are very old and I don't recommend them. Try building the current head of SmuView and use the current head of libsigrok as dependency. Both heads are (relatively) stable.

`sigrok::Quantity::ENERGY` is a libsigrok property that was introduced after 0.5.2, but SmuView relies on that property (as well as some other properties introduced after the 0.5.2 version).
Since some time (but after SmuView 0.0.5), SmuViews `cmake` step is checking if the used libsigrok version is new enough and would pint an error message otherwise.

Hope that helps.

TilCreator, 2022-12-25:
thx, I will try building with the newest commits of both. I'm planning on packaging this project for Nixpkgs, do you have any beta / semi stable version of both projects that could be used for that or should I wait for a new release of both before doing that?

knarfS, 2022-12-27:
As for SmuView, at the moment I'm currently using continuous releases/build, because "real" releases with a valid version number were very time consuming in the past and are arbitrary anyways.
That said, I'm planing to make a new "stable" release at the beginning of next year, but I wouldn't count on that ;)

I have no idea when a new libsigrok version will be released, but I wouldn't wait for that either. Just pick the latest head. libsigrok is pretty stable even when you use the latest git commit, but comes with a ton of new/improved drivers and bugfixes compared to the latest stable version 0.5.2

knarfS, 2023-02-25:
@TilCreator, I guess I can close this issue?

closed 2023-02-26 by TilCreator