knarfS/smuview #58: Wrong timestamp in CSV export with combined timestamp

SammysHP, 2023-12-09:
**Describe the bug**
When exporting data as CSV using "combine all timestamp", the written timestamp is wrong. It uses the timestamp from the highest resolution column even for columns with lower rate.

**To Reproduce**
Steps to reproduce the behavior:
1. Do a measurement
2. Export recorded data as CSV, using "combine all timestamps"

**Expected behavior**
Actual time of the measurements.

**Screenshots or log**
Here are two example files, saved from the same session. Output was enabled for approximately 30 seconds.

[combined_output_disabled.csv](https://github.com/knarfS/smuview/files/13624230/combined_output_disabled.csv)

[combined_output_enabled.csv](https://github.com/knarfS/smuview/files/13624231/combined_output_enabled.csv)

**Enviroment (please complete the following information):**
 - SmuView-0.0.6-git-a5ffb66-x86_64.AppImage
 - Arch Linux

closed 2026-11-06 by knarfS

knarfS, 2026-11-07:
Thanks for reporting.

The export function was completely reworked, but has a breaking change: The four header lines were replaced with only one header line, which now contains all information.

SammysHP, 2026-11-07:
Thanks a lot! I hadn't noticed that you started to work on this fantastic app again. Will check it out as time permits.